j7126/OctoPrint-Dashboard #2: Camera display.

SargonofAssyria, 2019-09-11:
Nice job! 
On my display there is enough room below the data to display the camera also.
Maybe as a setting to display the camera too on the dashboard.

StefanCohen, 2019-09-11:
Thanks. I'll look into that.

SrgntBallistic, 2019-09-11:
In the meantime you might want to check out the Webcam Sidebar plugin which will always display the plugin in the sidebar. Would probably have the same end result you're looking for

SargonofAssyria, 2019-09-11:
The problem with the Webcam Sidebar plugin is that I do not have enough room for all on the sidebar.

TalZiv, 2019-09-14:
first of all beautiful plugin.
Second, this is exactly what I wanted to post :)
Manged to this using Chrome dev mode, but would love to see it integrated with a smaller video window:
[image]

aherzberg, 2019-09-15:
I would also like to see this feature added. Would you be open to a pull request if I made the addition? Great job btw. Awesome Plugin.

StefanCohen, 2019-09-16:
I didn’t have time for this yesterday in 1.1.0 so I would appreciate a PR. I think this should have a new setting in the plugin settings (“Enable Camera”, disabled by default).

StefanCohen, 2019-09-17:
Basic support for a webcam has been added in nextrelease:

[image]

StefanCohen, 2019-09-18:
Fixed in 1.2.0

closed 2019-09-18 by StefanCohen

solyxius, 2019-09-19:
Great work Stefan, love the plugin!

With the new webcam view, would it be possible to a) "_shrink to fit_" (to eliminate the scroll bars), and b) enable a click to enter full screen view, another click to return to dashboard view (like the Sidebar Webcam plugin behaviour)?

[image]