j7126/OctoPrint-Dashboard #4: Control in dashboard

amsbr, 2019-09-11:
Nice job!

It would be interesting to have the controls on the dashboard (temperature, fan, etc.) using a single tab to see status and control.

Being able to control temperatures as it is on the temperature tab would make it easier to use a single tab.

StefanCohen, 2019-09-11:
Thanks for the feedback. I have considered this in the past few weeks but I have decided not to go the route of adding controls. At least not yet so I'll keep this suggestion active and wait for further response.

I want the dashboard to be as simple and informative as possible. So If I decide to go this route in the future, I'm thinking that it would be a setting to activelly enable controls.

chadlichty, 2019-09-19:
I really like your dashboard and also think it would be great to incorporate controls.  I do agree that the simplicity of the dashboard is what makes it so great.  I wonder if the gauges could double as the control. You have a line marking the set temperate.  You might be able to have that move based on mouse position on the gauge when clicked.

StefanCohen, 2019-09-20:
@chadlichty Nice idea but I don’t think I would be able to pull that off alone.

StefanCohen, 2019-09-21:
To understand the scope, I think it would be useful to have a list of all the controls you would like to see in the dashboard. Is it all of the controls provided by Octoprint today? More? Less?

chadlichty, 2019-09-21:
I would start with what you are currently showing. But yeah it would be cool to have control of everything. But I won't sacrifice the clean UI in order to add those controls.

StefanCohen, 2019-09-21:
So maybe something like a simple popup context menu on clicking the widgets?

Printer Profile - Select Profile
Connection status - Connect/Disconnect
Tool - Set temp from presets/any temp/off
Bed -  Set temp from presets/any temp/off
Job Progress - Cancel/Pause job

I think this could work and it wouldn't clutter the view.

chadlichty, 2019-09-23:
That could certainly do it.

StefanCohen, 2019-09-23:
I spent some time prototyping this over the weekend using [knockout.contextmenu ](https://github.com/nescalante/knockout.contextmenu). The bad news is that I totally failed to get it to work at all The good news is that I haven't given up yet :)

StefanCohen, 2019-09-27:
Update: I have managed to get the contextmenu to work and I have so far implemented Connect/Disconnect and Cancel/Pause. 

I will continue to play with this for a while and attempt to set the heaters based on the temp profiles from settings.

StefanCohen, 2019-09-29:
First version Added in v1.6.0. Additional context menues will be tracked using separate Issues, so closing this one.

closed 2019-09-29 by StefanCohen

chadlichty, 2019-10-02:
Very cool, nice work.