j7126/OctoPrint-Dashboard #5: Graph window too large.

Fribbler, 2019-09-11:
**Describe the bug**
The Graph window @ the temperature tab is too large for the layout.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Temperature window' 
2. Click on 'tab Temperature'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version; 76.0.3809.132

**Smartphone (please complete the following information):**
N.a.

**Additional context**
Cyborg theme installed. But with the plugin off I have the same problem.

StefanCohen, 2019-09-11:
Hmm. I haven't been able to reproduce this but I'll take a look at it again.

Fribbler, 2019-09-11:
Hi Stefan,

I have tried some more, and now I see that it is a problem in combination with Tempsgraph Plugin (0.3.4) If I disable this plugin the window is OK.

Fribbler, 2019-09-11:
Sorry, Wrong button.

StefanCohen, 2019-09-11:
Good find. I'm away from my dev env right now so I haven't analyzed this but there may be a CSS isssue with clashing Classes or IDs.

StefanCohen, 2019-09-15:
Strange. I have reproduced it. The tempsgraph Plugin sets up an SVG container that is 700*400 px when the dashboard is enabled but it is only 590* 400 when the dashboard is disabled. The standard Temperature tab is not affected. I can't find any css conflicts.
This may be a problem with the tempsgraph Plugin or the Dashboard, but I haven't been able to figure out what triggers the problem.

StefanCohen, 2019-09-20:
closing this issue since I don’t think it is caused by the dashboard after review

closed 2019-09-20 by StefanCohen