j7126/OctoPrint-Dashboard #6: Progress "Bar" Guage Obscures Percent Value

doughoyt, 2019-09-11:
When using "bar" progress gauge type, the bar obscures the actual percentage when you get to the ~ 50% mark.

Steps to reproduce the behavior:
1. Ensure progress gauge type is bar in plugin settings
2. Start print and observer that actual percentage displays
3. Wait until 50% or so and observer that percentage is obscured by bar
[image]

I would expect the percentage value to be "on top" of the progress bar and readable as the background changes during the bar's progress.

**Desktop (please complete the following information):**
 - OS: Win 10
 - Browser Firefox 
 - Version 69

Moving the class="progress-text-back" span _below_ the class="bar" div appears to fix the issue.  This is around line 101 of dashboard_tab.jinja2

[image]

StefanCohen, 2019-09-11:
Thanks. I didn't notice that myself. I'll try to reproduce it.

StefanCohen, 2019-09-15:
Fixed in 1.1.0

closed 2019-09-15 by StefanCohen