j7126/OctoPrint-Dashboard #14: Dashboard

iancattermole, 2019-09-13:
**Describe the bug**

Dashboard looks fantastic - just not on mine
I can't get more than the basic fan & temp to work on Windows laptop or Raspberry PI (3+ b)
I've installed "displayLayer" & "PrintTimeGenius" as recommended
I've tried uploading new code from Cura 4.1 and PrusaSlicer

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome

**Raspberry**
 - Device: PI Model 3B+
 - OS: Raspian Buster Lite
 - Browser Octoprint
 - Version 1.3.10

**Additional context**
I'm sure it's something I've done but being pointed in the right direction would really help and appreciated

Ender 3 Pro
Marlin 1.1.9
Raspberry PI 3 B+
Raspian Buster Lite
OctoPrint 1.3.10

Thanks

iancattermole, 2019-09-13:
"And make sure to indicate screen resolution and give an idea of how much screen space the browser is using. It looks like it's been shrunk down to me"

Raspberry PI 3B+
Raspian Buster Lite
Elecrow HDMI 5 inch 800*480 TFT (https://www.elecrow.com/wiki/index.php?title=HDMI_Interface_5_Inch_800x480_TFT_Display)

Lenovo 
lenovo ideapad s130-11igm
Windows 10
Octoprint Ver 1.3.10
1366 x 768

StefanCohen, 2019-09-13:
Thanks for the report. I'm away for a few days but will take a look when I'm back.

StefanCohen, 2019-09-15:
I haven't been able to reproduce this problem. I just want to make sure that this is the plugin you installed and not something else: https://plugins.octoprint.org/plugins/DisplayLayerProgress/

hoeni, 2019-09-15:
I have the same problem here. Output looks the same as in your picture.

StefanCohen, 2019-09-16:
@hoeni Do you have use a similar resolution as @iancattermole ? Do you have any errors in the browser console?

iancattermole, 2019-09-16:
Stefan, Really appreciate you looking into this but I am now sorted.
I've formatted and started again with the "stretch" version (https://octoprint.org/download/)
Clean install, same plugins, same theme, etc and it just works.....

I'm guessing I have been playing with so many different screens / drivers I had caused a problem. I now have an additional 0.5GB space so there was a lot of unused rubbish on there.

I'm sorted and thank you for your time. Ian

[image]

StefanCohen, 2019-09-16:
No problem. May I ask what version you upgraded from?

hoeni, 2019-09-16:
@StefanCohen I had different resolutions with no change (normal Safari and Chrome Browser on a huge screen).
However yesterday i f***ed up my OctoPi setup, so I'll have to reinstall. I'll come back, if I can (or can't) reproduce this.

iancattermole, 2019-09-16:
Yes, everything the same. Only difference was a load of drivers, etc no longer required and moving back to stretch from buster lite

hoeni, 2019-09-16:
So, up and running. The problem is absolutely independent of the browser size. Same in very small window as an Fullscreen on 5120 x 2880. Same in safari, chrome and firefox.
Octoprint 1.3.10 (OctoPi).

**The browser console says:**
Finalizing application startup packed_core.js:15521:17
Going to bind 33 view models... packed_core.js:15405:17
Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist packed_core.js:15458:37
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist packed_core.js:15458:37
Could not bind view model DashboardViewModel to target #tab_plugin_dashboard
<followed by a stacktrace of packed code, probably not helpful>

StefanCohen, 2019-09-17:
Ok. I need to figure out a way to reliably trace why the view model binding doesn't work for you.

StefanCohen, 2019-09-20:
Closing since I haven’t found a way to reproduce this.

closed 2019-09-20 by StefanCohen

hoeni, 2019-09-29:
After updating everything, it works fine now for me!