j7126/OctoPrint-Dashboard #15: Ability to turn on/off dashboard elements in setup

mcarlson-sb, 2019-09-14:
**Is your feature request related to a problem? Please describe.**
I currently do not have a fan, therefore I would like to turn off display of the fan speed dashboard element.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to be able to go into the Dashboard setup and toggle on/off display elements such as the fan speed

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
It might be possible to auto detect that fan input does not exist (this seems be to the case with a second extruder) however, a simple toggle would work well.  it would also allow people with a second extruder that isn't currently in use to be temporarily toggled to off.

**Additional context**
Add any other context or screenshots about the feature request here.

StefanCohen, 2019-09-14:
Thanks for the suggestion. This will require a setting in the plugin as I don't think OctoPrint provides a "hasFan" method.

StefanCohen, 2019-09-18:
I have added settings to enable/disable Fan, Octoprint Host System Info and Webcam for now. Chamber and second hotend are enabled/disabled in the OctoPrint Printer Profile settings.

StefanCohen, 2019-09-18:
Fixed in 1.2.0

closed 2019-09-18 by StefanCohen