j7126/OctoPrint-Dashboard #16: Layer count starts at 1 instead of 0

mstreurman, 2019-09-14:
**Describe the bug** the bug is.
When you start printing the layer count with the icon at the bottom left of the dashboard starts counting at 1 instead of 0 thus it always has 1 too many layers showing. The other plug in for it Displaylayerprogressplugin does show the correct count in the new window and on the printer

**To Reproduce**
Steps to reproduce the behavior:
Install dashboard 1.0
install Displaylayerprogressplugin
Slice object
Start printing
See the error

**Expected behavior**
To show the correct amount of layers

**Screenshots**
If applicable, add screenshots to help explain your problem.
N/A

**Desktop (please complete the following information):**
Intel Core i9 9900k
2x 8GB Corsair Vengeance LPX 3000MHz CL15
Asrock z390 Extreme4
nVidia GeForce RTX2080ti
250GB Samsung 960 EVO M.2 nVME
1TB Adata XPG SX8200 PRO M.2 nVME
Original Prusa i3 MK3S 3D-printer

Windows 10 Professional build 1903 running the latest updates for Windows and the latest drivers.
Edge browser (whatever is the latest version)

StefanCohen, 2019-09-14:
Thanks for the feedback.

This was intentional. Starting at zero only makes sense from a development perspective but I don't think it makes sense for the user if the first layer is layer zero.  I add 1 to the currentlayer and totallayers metrics from displaylayerprogress so for example a 10mm model sliced with .1mm layer height is reported as having a total of 100 layers (not 99) and the first layer is layer 1 (rather than 0). 

I could add a user setting in the plugin to let the user choose what method to use if enough people want it but I would still choose the current method (start at 1) as default.

This is my opinion but feel free to change my mind.

mstreurman, 2019-09-14:
[image]
The problem here is that if you have the displaylayerprogress plugin installed you will see conflicting information. eg. everywhere it shows layer 69 but in the dashboard it shows layer 70. That makes for a very inconsistent experience . That is: unless you will drop the dependency on a different plug in and make your own implementation of it or if the user manually changes all the settings of the displaylayerprogress plug in.

wizard04wsu, 2019-09-15:
Perhaps there should be better distinction between _completed_ layers and which layer is _currently printing_. Printing starts on layer 1, when 0 have been completed. Which number is the dashboard actually tracking?

StefanCohen, 2019-09-15:
I’m not sure that l understand. The dashboard shows the total number of layers in the gcode and what layer is currently being printed. Completed layers is not displayed.

wizard04wsu, 2019-09-15:
That makes sense to me. And the tooltip does say "Current Layer".

mkh595, 2019-09-16:
Addition 1 should be not fixed in code but as parameter in pluging setup, e.g. like in 'DisplaLayerProgress Plugin', there is explanation why

[image]

Model has 60 layers

[image]

but in Dashboard are both values wrong due to fixed 'addition 1'

[image]

StefanCohen, 2019-09-16:
Thanks! I wasn't even aware of that settin in DisplayLayerProgress. I will remove the layer increment in the dashboard and update the README with the info.

StefanCohen, 2019-09-16:
Fixed in next release.

StefanCohen, 2019-09-18:
Fixed in 1.2.0

closed 2019-09-18 by StefanCohen