j7126/OctoPrint-Dashboard #17: Icons not loading - Using Static Paths instead of relative?

dbird86, 2019-09-15:
The Dashboard Plug-in is attempting to display the images from: http://<RPI-IPAddress>/plugin/dashboard/static/img/printer-icon.png

Problem is I run three instances of OctoPrint on my RPi so each one sits under a "sub-site" for example: http://<RPI-IPAddress>/printer1/ and http://<RPI-IPAddress>/printer2/, etc

So is it possible to update the Plug-in so that it uses relative paths and looks in http://<RPI-IPAddress>/<printername>/plugin/dashboard/static/img/ for the images

StefanCohen, 2019-09-15:
Thanks for the feedback. The URLs for bundled static images are automatically generated by the plugin framework so I’m not sure this is something i can address.

StefanCohen, 2019-09-15:
I have researched this now and I can’t find a way to do it that supports all installation methods (plugin manager, pip, setup.py devtools). I’m closing it for now but feel free to reopen if you or anybody else can find a way to do it.

closed 2019-09-15 by StefanCohen

dshokouhi, 2019-09-16:
@StefanCohen can we maybe specify a baseurl if we are using it so works properly?  I am using nginx and serving octoprint over /octoprint but only the icons are served up under the domain name and not the sub folder that i set.  Maybe its a configuration setting on my end?  Its just odd how the custom icons work when i use themify but not in this plugin.