j7126/OctoPrint-Dashboard #20: Feature Request: Raspberry Pi Performance Specs

SrgntBallistic, 2019-09-16:
I've been pretty surprised to see that there's no active plugins to view a Raspberry Pi (or other machine) system performance specs. Like CPU Temp/Speed/Utilization, Ram Utilization, Storage, Etc...

This would be particularly useful for Pi's. I've been using Octolapse and playing around with using a Wyze Cam as a camera source. This along with the other many plugins has me concerned with the Pi throttling and impacting prints.

I started designing/wireframing a possible tab to show these specs and... Voila your plugin popped up. Seems like it might be more worthwhile to try and add this to your plugin than make it a standalone plugin.

There's an older plugin that doesn't seem to work anymore. I was going to use this as a starting point:
https://github.com/myersCody/OctoPrint-Cpustatus

Looks like they were basically writing the results of psutil to a text file and reading that into the UI.

`cpuUsage = psutil.cpu_percent(interval=1)
testFile = open("test.txt", "w")
testFile.write(cpuUsage)
testFile.close()`

I believe the author of that plugin pull some code from NavBarTemp. Looking at that he has a class for getting the system temp for different types of systems including Raspberry Pi's.

https://github.com/imrahil/OctoPrint-NavbarTemp/blob/master/octoprint_navbartemp/libs/sbc.py

StefanCohen, 2019-09-16:
That sounds interesting and would be a nice fit I think. Would you like to collaborate on that?

SrgntBallistic, 2019-09-16:
> That sounds interesting and would be a nice fit I think. Would you like to collaborate on that?

Quite possibly. I've never worked on an OctoPrint Plugin or contributed to a github for that matter. But I think I have the capacity to do so.

What would you recommend to do to start?

StefanCohen, 2019-09-16:
> Quite possibly. I've never worked on an OctoPrint Plugin or contributed to a github for that matter. 
That makes two of us :)

I looked at psutil. It should be possible to include it in the plugin and have it installed at the same time using the plugin manager. I don't see why it would be necesary to write it to file unless you want time graphs.

What metrics would you like to include? Given the grid-based layout of the dasboard, I'm thinking multiples of three. Something like: CPU usage, Mem usage, Disk Usage. 
After that: CPU Temp, Load, ???

Can you upload a wireframe of what you had in mind?

StefanCohen, 2019-09-16:
Getting the CPU, MEM and HD/SD info via psutil was easy:
[image]

Is this similar to what you had in mind?

SrgntBallistic, 2019-09-16:
That's a pretty good turn around! Hahah. That's pretty much exactly what I had imaged and started writing up. I had names written out instead of icons but those seem perfect. Temp is really the only thing I can think of that would be useful at the moment. I'm sure the community might be able to suggest some others to keep the 3 column grid going.

You Rock!

StefanCohen, 2019-09-16:
I added the CPU temp. That may unfortunately not work on anything other than an RPi as the sensor ID is hardcoded for now. I'll see if I can address that before the next release.
[image]

StefanCohen, 2019-09-18:
Fixed in 1.2.0

closed 2019-09-18 by StefanCohen

vbz89, 2019-09-27:
Does this work for any RPi? I have a RPi 3B and I get "0" Values for all 3 parameters.

[image]

StefanCohen, 2019-09-27:
It *should* work on any RPi and it does work on my 3b and 3b+. Are you on OctopPi? What version in that case?

it looks like psutil doesn't return anything.

What happens if you try the following from a terminal:
`source ~/oprint/bin/activate`
`python`
`>>>import psutil`
`>>>psutil.virtual_memory().percent`

vbz89, 2019-10-01:
Sorry for the late reply, I am using octopi 0.14.0, octoprint 1.3.11

The above commands gave me "32.2"

StefanCohen, 2019-10-01:
Ok. That means that psUtil is at least available. I wonder if there may be a difference between OctoPi 1.14 and 1.16 that I use? Do you have the possibility to try a later version?

vbz89, 2019-10-01:
I will take a backup image of my sd card and try a fresh installation tomorrow and get back to you on the results on 0.16. I appreciate the quick response. Thanks.

vbz89, 2019-10-02:
I tried today on a clean install of 0.16, it is working correctly. So I guess it's a problem with 0.14. Time to shift all my stuff to a new installation I guess.

StefanCohen, 2019-10-02:
Ok. Good to know. I will set a minimum version in the installer

o5, 2020-09-20:
I've installed this plugin and enabled "Show RPi System Info" and it does not work :/ 

[image]

```
Dashboard 1.14.0
OctoPrint 1.4.2
Python 3.7.3
Raspbian 5.4.51-v7l+ #1333
Raspberry Pi 4 Model B Rev 1.1
```

I tried to install psutil using pip and it looks work
```bash
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> psutil.virtual_memory().percent
14.7
>>>
```