j7126/OctoPrint-Dashboard #21: Fan speed missing

zebons, 2019-09-16:
I installed your nice dashboard plugin.
However a got an issue, Fan speed gauge always stay at zero and don't show the real Fan speed value (value is correctly showed on the control box). On a CR-10S with the E3Dv6 creality hotend (2 fans) and firmware marlin 1.1.9.

zebons, 2019-09-17:
Yes strange. I use S3D as slicer and you’ll see in the gcode file joined that Fan are controlled by M106 commands :
M106 S0 for the first layer
M106 128 from layer 2

I tried to send M106 commands via the Octoprint Terminal window but same issue : Fan speed really change and correctly displayed on the control box but display remain to OFF in the dashboard window.

I’m using  the Fan Speed Control plugin and was thinking that it could be the origin of my issue so I disabled it and send M106 commands via Octoprint terminal but same behaviour (Fan speed remains to OFF in the dashboard window)

From: StefanCohen
Sent: Monday, September 16, 2019 8:20 PM
To: StefanCohen/OctoPrint-Dashboard
Cc: zebons ; Author
Subject: Re: [StefanCohen/OctoPrint-Dashboard] Fan speed missing (#21)

Strange. I will try to reproduce it. What slicer are you using? Can you upload the gcode here?

StefanCohen, 2019-09-17:
Ok. I have been trying to reproduce this for a while now without success. Can you please upload a screenshot and any errors from the browser console.

ProtomakerSprint, 2019-09-17:
I have the same issue, this is not a complaint but hope to help you with a great plugin Fan sppen not dispalyed

Mine is set to 50% fan speed: Gcode M106 S127, Turn on at Layer 3 for PLA

Image screen shoot note latest plugin 17 September 2019

**
[image]
**

StefanCohen, 2019-09-17:
Hi. I see that you don’t have the DisplayLayerProgress plugin installed. Installing it as per the message at the bottom of the Dashboard should fix this. There is more info in the README. @zebons Is this the case for you too?

zebons, 2019-09-17:
Yes, I used the DisplayLayerProgress plugin earlier but had to deactivate as it gave me an issue with a previous version of Cura (pause during prints). I reactivate it to do some tests but I can get the latest version as my Python is out of date (needed python 3.5 ans I have 2.7 installed). So I have to reinstall octoprint with the latest version to have python 3.5 installed. As soon as I have done, I'll test with the DisplayLayerProgress plugin  installed.

StefanCohen, 2019-09-17:
A disabled displaylayerprogress explains the issue. But why would you need Python3? Octoprint only supports 2.7.

zebons, 2019-09-17:
I confirm that I have the Fan speed with the DisplayLayerProgree plugin ionstalled and activated.
For Python it' the errot that I get when I try to update some plugin (Printer Stat 2.0.0 for ex) : 
```
2019-09-17 18:38:13,727 > File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
2019-09-17 18:38:13,729 > six.reraise(type, exc, self._tb)
2019-09-17 18:38:13,730 > File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
2019-09-17 18:38:13,732 > yield saved
2019-09-17 18:38:13,734 > File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
2019-09-17 18:38:13,735 > yield
2019-09-17 18:38:13,737 > File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
2019-09-17 18:38:13,739 > _execfile(setup_script, ns)
2019-09-17 18:38:13,740 > File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
2019-09-17 18:38:13,742 > exec(code, globals, locals)
2019-09-17 18:38:13,743 > File "/tmp/easy_install-2yhGLo/numpy-1.17.2/setup.py", line 31, in <module>
2019-09-17 18:38:13,745 >
2019-09-17 18:38:13,747 > RuntimeError: Python version >= 3.5 required.
2019-09-17 18:38:13,748 >
2019-09-17 18:38:13,750 > ----------------------------------------
2019-09-17 18:38:13,754 ! Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-4Nfc5Q/pandas/

```

StefanCohen, 2019-09-17:
Ok. I’m glad it worked. I have created a feature request to make it more obvious that displaylayerprogress is required. I can’t help you with the Python issue. I suggest that you turn to the forums for help. @ProtomakerSprint Does this solve your issue too?

zebons, 2019-09-17:
ok the python issue is specific to the Printer Stat plugin update so you can close this thread for my need.

StefanCohen, 2019-09-17:
Ok. Closing this issue as I now see from @ProtomakerSprint screenshot that he too is missing the DisplayLayerProgress plugin.

closed 2019-09-17 by StefanCohen

anubisg1, 2022-01-14:
Is this dependency still existing? I don't have installed display layer progress plugin and I have the same problem.

The dependency, if there seems to be undocumented, if not there then this issue still exists today.

j7126, 2022-01-14:
> Is this dependency still existing? I don't have installed display layer progress plugin and I have the same problem.
> 
> The dependency, if there seems to be undocumented, if not there then this issue still exists today.

Hello @anubisg1, this dependency no longer exists. As of version 1.19.0, DisplayLayerProgress is no longer required.

In regards to you still having the issue, please let me know what versions of dashboard and OctoPrint you have installed, and please upload your gcode file so that I may debug the issue.

anubisg1, 2022-01-15:
Hello @j7126

The issue seems to exist only if the print is initiated from the SD card. Either manually from the printer or via octoprint.

If the gcode is stored on octoprint and streamed via serial port, the fan diagram shows correctly. Everything else works perfectly except for the fan

j7126, 2022-01-26:
> Hello @j7126
> 
> The issue seems to exist only if the print is initiated from the SD card. Either manually from the printer or via octoprint.
> 
> If the gcode is stored on octoprint and streamed via serial port, the fan diagram shows correctly. Everything else works perfectly except for the fan

Hi @anubisg1,

This is expected when printing via the sd card, this informtion is only available when streaming gcode through OctoPrint. 

The reason for this is that when printing from the printer sd card, there is not any way for the plugin to see and analyze the gcode in order to get this information. As you can see in your screenshot, the layer height and layer time info down the bottom is also missing as well as the fan speed, and this is expected behavior when printing from sd card. 

I don't know of any way to get this information when printing form the sd card.

j7126, 2022-01-26:
Actually I now realize that marlin does have the ability to report the current fan speed, so there should be a way to do this for the fan speed.

j7126, 2022-01-26:
Alright I have added the ability to get the fan speed when printing from sd card and when the fan speed is changed from the printer LCD. 

This feature will be included in the next release. It will need REPORT_FAN_CHANGE enabled in your printer firmware to work.

j7126, 2022-01-26:
@anubisg1 you can test this out now if you want by installing the development branch of the plugin by using the url below in the plugin manager. As long as you have REPORT_FAN_CHANGE enabled in your firmware, the fan speed should show up.

`https://github.com/j7126/OctoPrint-Dashboard/archive/refs/heads/development.zip`

anubisg1, 2022-01-26:
Absolutely, let me test it today

anubisg1, 2022-01-27:
i have not forgotten, i just have a very long print on going. should finish in the next 6 hours, after that i will be able to safely install the plugin update

anubisg1, 2022-01-27:
> you want by installing the development branch of the plugin by using the url below in the plugin manager. As long as you have REPORT_FAN_CHANGE enabled in your firmware, the fan speed should show up.

It works!