j7126/OctoPrint-Dashboard #23: Temperature Gauge Maximums settings do not work for Bed Max

THHL, 2019-09-17:
The Temperature Gauge Maximums settings for Bed Max has no effect.
(The same setting for Hotend Max works as expected.)

Steps to reproduce the behavior:
1. Go to Dashboard Plugin Settings in OctoPrint
2. Click on Bed Max button down
3. Save settings
4. In Dashboard, see that the Gauge did not change

**Expected behavior**
The Gauge should have changed according to the new Bed Max setting

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 17.04
 - Browser: Firefox
 - Version: 69.0

StefanCohen, 2019-09-17:
Reproduced. Will fix

StefanCohen, 2019-09-18:
The template from the original PR got overwritten at some point so the changes got lost after testing. I have added the correct version to the next release.

StefanCohen, 2019-09-18:
Fixed in 1.2.0

closed 2019-09-18 by StefanCohen