j7126/OctoPrint-Dashboard #25: Progress reported is incorrect

jimmyeao, 2019-09-18:
**Describe the bug**
A clear and concise description of what the bug is.
The dashboard (and now OCtoprint) is reporting incorrect progress that conflicts with the LCD progress and real progress

**To Reproduce**
Steps to reproduce the behavior:
Start any print

**Expected behaviour**
Progress should be correct - see attached, reports over 40% complete on 2nd layer. Printer reports 14%

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win 10
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22]  76.0.3809.132

**Additional context**
Add any other context about the problem here.

jimmyeao, 2019-09-18:
Its the same, so perhaps something weird going on with Octoprint?

StefanCohen, 2019-09-18:
Yes. This is likelly something with OctoPrint so I'll close this issue. You can also try installing OctoPrint-PrintTimeGenius to see if that helps.

closed 2019-09-18 by StefanCohen