j7126/OctoPrint-Dashboard #26: Autohide Hotend display if not used

LMS0815, 2019-09-18:
My printer has 2 hotends, but only 1 is used at a time. 
To save some space I would like to (auto)hide the display of a hotend, if the target temperature is = 0 (zero). 
This would free space for the Fan Gauge (which I had to hide because of limited space.
[image]

StefanCohen, 2019-09-18:
I think we can try this as an "advanced" setting.

StefanCohen, 2019-09-20:
Allright. I have tried this but I'm not sure that this is good enough. The new setting "Hide hotend(s) when target temp = 0" will hide _all_ hotends until at least one is heated up. This will in your case mean that only the bed and fan is visible when the printer is idle. Is that what you was looking for?

StefanCohen, 2019-09-20:
Fixed in 1.3.0

LMS0815, 2019-09-20:
Yes. Perfect. If thee is no Temperatur set there is no need to monitor. I have to check it,but I am very sure I'll Ike it. THANK YOU!

StefanCohen, 2019-09-20:
The feature was causing problems for others so I had to revert it and push 1.3.1 without it. I will have to give it a go some other day.

StefanCohen, 2019-09-21:
Allright. I think I have fixed it now. I will release it this weekend.

closed 2019-09-23 by StefanCohen