j7126/OctoPrint-Dashboard #30: Feature Request: Orange PI Performance Specs

alakond, 2019-09-19:
You have released a plug-in for Rasbery PI, is it possible to add support for Orange PI? Thnx!
[image]

StefanCohen, 2019-09-19:
Thanks for the feedback.
I don't have access to an Orange PI, so it will not be possible for me to support it. But feel free to submit a pull request and I will take a look at it.
The Fan, layer, height and avg layer time issue is related to the warning message at the bottom of the Dasboard in your screenshot.

alakond, 2019-09-19:
Thanks for the answer.
I can give remote access to my orange pi SSH and the web to my octoprint.

StefanCohen, 2019-09-19:
Thanks, but that is not a long-term solution for me.

closed 2019-09-19 by StefanCohen