j7126/OctoPrint-Dashboard #32: Web Socket Causes Page to Never Finish Loading

ShodaPublicity, 2019-09-20:
The new changes to enable/show the webcam on the dashboard causes a web socket connection to never finish. In Firefox this causes the webpage to continuously show that it is loading in the tab. In Chrome it says it finishes loading but if you look at the network tools you will see the websocket for webcam streaming never finish loading/connecting.

Not sure what information you need but it should be easily recreated. If you need any information from me let me know.

Chrome version:  77.0.3865.90 
Firefox version: 69.0.1

StefanCohen, 2019-09-20:
Is it a remote camera or USB?

ShodaPublicity, 2019-09-20:
It is a USB camera.

StefanCohen, 2019-09-20:
Except the connection that is required to stay up for the webcam stream, I can’t see anything suspicious in my environment.

ShodaPublicity, 2019-09-20:
Ah sorry, I thought it was the webcam because I enabled that and then started seeing the issue, but I forgot that I actually enabled "Show Octoprint Host System Info" first. 

I disabled the host system info and that fixed the issue so the issue is actually due to that.

StefanCohen, 2019-09-20:
No problem. Thanks for the feedback.

closed 2019-09-20 by StefanCohen