j7126/OctoPrint-Dashboard #33: Version 1.3.0 hides the extruder temp, even when printing

AdrianCarreno, 2019-09-20:
**Describe the bug**
Today I updated to version 1.3.0 and the extruder gauge is missing. I went into the settings and noticed a checkbox to hide the gauge whent target temp is set at 0, but it is unchecked, and the gauge is missing no matter what target temp is set (maybe the problem is unrelated to that checkbox).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrome
 - Version 76.0.3809.132

**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi A2
 - OS: Android 9
 - Browser Chrome

**Additional context**
Serer running on OctoPi

StefanCohen, 2019-09-20:
Confirmed bug. I have removed the feature and published 1.3.1. Please confirm if the issue has been resolved.

slm4996, 2019-09-20:
I can confirm that 1.3.1 fixes the missing extruder widgets.

closed 2019-09-20 by StefanCohen