j7126/OctoPrint-Dashboard #34: broken links behind reverse proxy

tombic, 2019-09-20:
Please make links to icons relative for octopi installations behind simple reverse proxy. 
When octopi is behind reverse proxy and in subdirectory it can't find links to icons, because address is starting with /. If it is possible can you make links relative to page (with no / at the beginning of address?) For example "/plugin/dashboard/static/img/hotend-icon.png" -> "plugin/dashboard/static/img/hotend-icon.png".
BTW, very nice plugin. It is helpful to view all the status info on one place.

StefanCohen, 2019-09-20:
It seems like the slash in the beginning has to be there in order to work on a non-proxied installation. Have you tried removing it locally to verify that it still works in both scenarios?

tombic, 2019-09-20:
When checked locally (changed that in firefox inspection window) it works without slash in both proxied and non proxied scenario. But I can't say that it may not create any problem somewhere else as I am not web developer.

StefanCohen, 2019-09-20:
> I am not web developer.

Me neither. I’m just pretending :)  Ok. I will have to test all the install methods to make sure it doesn’t breaks something.

tombic, 2019-09-23:
1.4.0 works for me locally and over internet and proxy too. Good.
You pretend very well :-))) Thanks.

closed 2019-09-23 by tombic