j7126/OctoPrint-Dashboard #37: "Auto scroll" webcam -> follow current layer

LMS0815, 2019-09-21:
Please add an "Auto scroll" (and zoom?!) option as an alternative to scaling the webcam picture.

The idea is to display a frame of the picture (without scaling and or zooming) which is aligned with the current layer. This would enable the frame to follow the nozzle. (This is what I need to see in the dashboard). A full view is elsewhere available.

As we know the layer height (e.g. 0.2 ) we only need to know two positions and the value of those within the webcam picture ( minimum height on display - usually 0 mm and maximum height usually the y-max in mm). 

Please think about this - it sounds more complicate than it is.

StefanCohen, 2019-09-21:
I don't have the slightest idea how to do that but please provide a pull request so I can evaluate the suggestion.

LMS0815, 2019-09-21:
Can be done easily with CSS:

See:
https://stackoverflow.com/questions/57725/how-can-i-display-just-a-portion-of-an-image-in-html-css

and

http://jsfiddle.net/davidThomas/ygds2eeu/

StefanCohen, 2019-09-21:
Ok. I will as I said take a look if you want to contribute a pull request.

LMS0815, 2019-09-21:
I am not able coding python. what shall I contribute within  a pull request? (Never made one before).

closed 2019-09-21 by LMS0815