j7126/OctoPrint-Dashboard #38: Full screen mode

Ha-Z-Z-ard, 2019-09-21:
**Is your feature request related to a problem? Please describe.**
no problem,is it possable to have the dashboard within its own frame,so this could be made fullscreen?

**Describe the solution you'd like**
fullscreen dashboard for a " info screen full display" with setups that have screens for there printers

**Describe alternatives you've considered**
just learning code and alterntives are beyond my current expertise

StefanCohen, 2019-09-21:
Ok. I have added a setting for enabling fullscreen mode (experimental for now) in the next release. I haven't made any attempt to style it differently if in fullscreen yet. I think the layout should be adjusted significantly when in fullscreen, but my css-fu is very limited, so let's see if you or anybody else would like to contribute some styling.

Ha-Z-Z-ard, 2019-09-21:
Amazing Stefan, thank you

I have limited knowledge with css styling but wouldn't mind learning so
will have a wee prod about

Scott Reid

chadlichty, 2019-09-21:
This is great.  One issue I saw was the webCam was cropped in full-screen mode.  That issue is fixed in the regular display.  Looks great though.

StefanCohen, 2019-09-21:
I know. The webcam issue is however technically different than the one in 1.2.0 and can be solved by restyling the view in fullscreen.  
The fullscreen mode could benefit from a different layout where the cam and dash resides side by side horizontally or maybe overlaid on eachother.

Ha-Z-Z-ard, 2019-09-21:
This is amazing..... How easy would it be to move the gcode viewer side by
side with webcam in y. the full screen mode?

I think it. Would be amazing to have all the stats at the top,webcam and.
Gcode viewer Centre then progress at time finish at bottom.

Scott Reid

StefanCohen, 2019-09-21:
It is certainly possible to combine the views.Consolidate Temp Control does something similar: https://github.com/jneilliii/OctoPrint-ConsolidateTempControl

But I'm not sure I want to go that route. I think Less is more and I would rather do something like FullScreen WebCam does and as you mentioned show the webcam view fullscreen and overlay a bar or two with the gauges insted: https://plugins.octoprint.org/plugins/fullscreen/ I think I'll mock it up this weekend to see what makes sense.

Ha-Z-Z-ard, 2019-09-21:
Sounds great, I think your work is great so whatever you want to do is great

Scott Reid

StefanCohen, 2019-09-21:
Maybe something like this:

[image]

Ha-Z-Z-ard, 2019-09-21:
Dude, that looks amazing.

Scott Reid

xeddog, 2019-09-22:
Just a suggestion, but I would much rather see full screen WITHIN THE BROWSER Window.  That would allow me to scale browser windows and see two or more different instances on a single screen.

chadlichty, 2019-09-22:
Looks great!

StefanCohen, 2019-09-22:
@xeddog I think I like that better than the current fullscreen. It is a bit like Octoprint TouchUI works. 

Maybe something like this?

[image]

j7126, 2019-09-22:
That looks pretty good. I would be happy to give any help if you need.

StefanCohen, 2019-09-22:
@j7126 I would appreciate any help I can get with this. CSS is pretty much black magic to me. 

The Idea I have is to go with the layout above (dashboard left, webcam right) when In fullscreen. That would (hopefully) not need a separate jinja template. I would like to keep the rest of the styling   and compatibility with Themeify but I like your approach of placing the icons above the texts as you did with the TouchUI PR.  

Maybe something like this, but I'm open for better alternatives: 
[image]

The question is what to do if there is no Webcam or if it is disabled. One approach could be to just center the dashboard and spread it out a bit horisontally. Do you have any suggestions?

I would really appreciate if you would like to collaborate on this.

j7126, 2019-09-22:
I will try to do something like that, the only thing is I do not have a webcam to test with but I will try to work around that.

StefanCohen, 2019-09-22:
Cool. I guess you can link to any publicly available webcam feed from a site such as opentopia or maybe just a static image. I just tried and here's one that works: http://92.60.192.245/cgi-bin/faststream.jpg?stream=full&fps=0

jimmyeao, 2019-09-22:
> @xeddog I think I like that better than the current fullscreen. It is a bit like Octoprint TouchUI works.
> 
> Maybe something like this?
> 
> [image]

That looks awesome!

closed 2019-09-23 by StefanCohen