j7126/OctoPrint-Dashboard #39: Webcam not showing on dashboard (in 1.4.0).

JonS9999, 2019-09-22:
**Describe the bug**
I upgraded from 1.3.x to 1.4.0, and now the webcam is no longer being shown on the dashboard tab.

**To Reproduce**
Steps to reproduce the behavior:
1.  Install 1.4.0.
2.  Configure Dashboard to display the webcam.
3.  Observe that the webcam is not being displayed on the Dashboard tab.

**Expected behavior**
When enabled, the webcam should be displayed on the Dashboard tab.

Running on Octopi.
Viewing in Chrome (Windows 7 64-bit).

StefanCohen, 2019-09-22:
Ok. Is it related to this: https://community.octoprint.org/t/dashboard-plugin-feedback-needed/11758/51?u=stefan_cohen

What do you have in the Stream URL in the Webcam & Timelapse settings?

michi-e, 2019-09-22:
Hi,
just discovered the same behavior starting with version 1.3.0 and and wanted to give some additional information that might help tracking down the problem.

At least in my case, the bug manifests when there are other plugins active that relocate the webcam view--tried with 'Webcam Tab' and 'TouchUI' both of which move the webcam to a separate tab (and dont't interfere with one another).

After playing around a little, I could observe the following behaviours:
1. "Webcam Tab" enabled: Webcam appears (working) in own tab but not on 'Dashboard'; 'Controls' tab shows the webcam field (which it souldn't) with an error message.
2. "TouchUI" enabled:
2.1. Standard view: Webcam appears on both 'Dashboard' and 'Control' tabs and works as expected.
2.2 TouchUI view: Webcam appears  neither on the 'Dashboard' nor on the separate 'Webcam' tab; 'Controls' tab interface is completely broken and shows the webcam field with an error message as background.
3. Both "Webcam Tab" and "TouchUI" enabled:
3.1 Standard view: See 1.
3.2 TouchUI view: See 2.2

StefanCohen, 2019-09-22:
Allright. 
I think this points to a bigger problem that may be hard to address for plugin developers. The Control tab tries to save cpu cycles by shutting down the webcam stream whenever the tab is unselected for more than 5 seconds. If you want to display the stream anywhere else, you basically have to trick the control tab to beleive that it is currently active. I think plugins that are not aware of each-other may get into conflict in manipulating the control tab. I don't see any solution to this but will ask around. I have possibly got something wrong.

I see two short-term solutions for the dashboard 
1. Define a separate webcam config and ignore the one defined in Octoprint. 
2. Remove the webcam support.

StefanCohen, 2019-09-23:
I have reviewed how the dashboard works with the other plugins that displays the webcam and some of them re-writes internal octoprint functions to a level where it is no longer possible for the dashboard to access the stream. I have written up the longer version over at the [wiki](https://github.com/StefanCohen/OctoPrint-Dashboard/wiki/Help#webcam-access-issues) 

I will probably have to implement a standalone webcam config to get around this. The short-term solution is for users to disable conflicting plugins.

closed 2019-09-23 by StefanCohen

JonS9999, 2019-09-23:
Hi Stefan.

Thanks for looking into this.

Just as a data point (and to reiterate my original post):
- I am always able to see the webcam (on the Dashboard) using Dashboard version 1.2.0.  
- I am never able to see the webcam (on the Dashboard) using 1.3.0 or higher.

I appreciate what you are saying about having to trick OctoPrint into thinking it's on the control tab, etc, however, given what I'm seeing, it implies that something changed between 1.2.0 and 1.3.0 that is causing this issue.

Does this make sense?

-Jon

JonS9999, 2019-09-23:
> Ok. Is it related to this: https://community.octoprint.org/t/dashboard-plugin-feedback-needed/11758/51?u=stefan_cohen
> 
> What do you have in the Stream URL in the Webcam & Timelapse settings?

My "Stream URL" field says:   /webcam/?action=stream

Cheers.

-Jon

[image]

StefanCohen, 2019-09-23:
Yes. The wecam view changed completely from a really simple iframe in 1.2.0 to a copy of the Control tab view in 1.3.0. v1.3.0 and later is thus more vulnerable to the problems i mentioned earlier but it also supports all the other webcam settings that Octoprint provides (scaling, rotation, flipping, aspect ratio, etc) without having to double the configuration. (I'm honestly not sure it is worth the tradeoff) 

I can come to think of two potential problems that you may experience.

1. You may have a conflicting plugin installed. Try to temporarilly disable all other plugins except the dashboard.
2. Try upgrading to 1.5.0 The fix to #44 could be a solution if you're behind a reverse proxy for example.