j7126/OctoPrint-Dashboard #40: Colors for temperature graphs

xeddog, 2019-09-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
No problem
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Color changing temperature graphs for nozzle and bed temps.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Are there any????
**Additional context**
Add any other context or screenshots about the feature request here.
I think I would like to see color changing temperature graphs for nozzle and bed temps.  Something along the lines of :
Blue when the temperature is initially set and up until the next stage
Green when the temperature is + or - 2 degrees of target temp.
RED when the temperature is + or - 5(?) degrees from the set temperature after reaching set temp and stabliizing

This could potentially show the need for running PID tune, or a potential problem coming up, etc. with just a couple of glances at the dashboard.

StefanCohen, 2019-09-22:
I had exactly this in mind early on and I still think it is a good idea. Take a look here:  https://community.octoprint.org/t/dashboard-plugin/11536/46?u=stefan_cohen

I might give it a try after I've sorted out the fullscreen layout.

Naugrimohtar, 2019-10-03:
How about similar for the CPU temp meaning change when getting high?

StefanCohen, 2019-10-06:
@Naugrimohtar I like that but how do we define what "High" is for the CPU? Are there themal limits for the RPi specified somewhere that we can use or do we need a user setting for this?

xeddog, 2019-10-06:
Maybe this will help.  From https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md

"For Raspberry Pi 3 Model B+, the PCB technology has been changed to provide better heat dissipation and increased thermal mass. In addition, a soft temperature limit has been introduced, with the goal of maximising the time for which a device can "sprint" before reaching the hard limit at 85'C. When the soft limit is reached, the clock speed is reduced from 1.4GHz to 1.2GHz, and the operating voltage is reduced slightly. This reduces the rate of temperature increase: we trade a short period at 1.4GHz for a longer period at 1.2GHz. By default, the soft limit is 60°C, and this can be changed via the temp_soft_limit setting in config.txt."

Naugrimohtar, 2019-10-06:
Agreed! I googled briefly and saw that the RPi 3 scales speed at 80C core
temp. Also read that the 3b+ scales as low as 70c. I also saw and article
thy detail was changed to 60c but could be overridden (
https://github.com/raspberrypi/firmware/issues/1014#issuecomment-401599394
)

Perhaps user settable with 57c yellow and 65c red to start? I wonder if
there is a way to query the OS for throttle temps?

StefanCohen, 2019-10-07:
Keep in mind that OctoPrint might just as well be running on a Xeon instead of ARM so we will need user settings for the CPU thresholds but might set them to something like 60 for yellow and 80 for red by default regardless. 

The hotend and bed thresholds are different and should probably behave like @xeddog described above.

xeddog, 2019-10-08:
I didn't think about that.  All the different processors would add a few thousand wrinkles, wouldn't it.

Naugrimohtar, 2019-10-08:
Wrinkles yes but those with other platforms might not care to monitor their core temp so perhaps in addition to being able to specify the temps, we could slow enabling:disabling of that ‘widget’, and the other widgets too?

xeddog, 2019-10-09:
But if they did care to, then the author could potentially be swamped with detailing the temperature characteristics of hundreds of processors.  Maybe an option might be to let you specify your own temperatures.

LMS0815, 2019-10-09:
What's about providing a user defined field for CPU?
And stick to the original idea.

> Blue when the temperature is initially set and up until the next stage
> Green when the temperature is + or - 2 degrees of target temp.
> RED when the temperature is + (or - 5? I would regard low as OK...)

StefanCohen, 2019-10-12:
WIP for 1.11.0:

[image]
[image]
[image]

Will work on the temp gauges next.

StefanCohen, 2019-10-12:
I added the cpu frequency to the CPU widget as that is relevant to the temp reading and throttling:

[image]

StefanCohen, 2019-10-12:
WIP:

[image]
[image]

There is now a setting to enable the temp gauge coloring (default off). There is also a temp deviation setting (default 10C). It works like this:

* Blue if target = 0
* Blue below target - deviation
* Green between target - deviation and target + deviation
* Red above target + deviation.

Naugrimohtar, 2019-10-13:
Excellent!

StefanCohen, 2019-10-13:
Fixed in 1.11.0

closed 2019-10-13 by StefanCohen