j7126/OctoPrint-Dashboard #41: Hi res webcam image doesnt scale well

jimmyeao, 2019-09-22:
**Describe the bug**
Camera has bottom half cut off in full screen mode

**To Reproduce**
Steps to reproduce the behavior:
Configure camera for 1280x720, enter full screen mode, only top half of image is shown

**Expected behavior**
Image would be scaled to fit

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win 10
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22] Current Version

**Additional context**
Love the full screen view. Would be good to have a 'panic' button to cancel the print. Maybe one we can add some custom gcode to so we can say, stop print and move up 5 mm

StefanCohen, 2019-09-22:
Thanks. I'm aware. That's why the fullscreen option is marked "experimental". The fullscreen CSS will be redesigned.

j7126, 2019-09-23:
Fixed

StefanCohen, 2019-09-23:
Fixed in 1.5.0

closed 2019-09-23 by StefanCohen