j7126/OctoPrint-Dashboard #46: Conflict with WebcamTab

jneilliii, 2019-09-24:
**Describe the bug**
When both the Dahboard plugin and the WebcamTab plugin are installed the WebcamTab plugin no longer works.

**To Reproduce**
Steps to reproduce the behavior:
Install both plugins and you'll see webcamtab no longer shows video.

**Expected behavior**
For both plugins to work.

**Additional context**
As reported over [here](https://github.com/foosel/OctoPrint/issues/3281).

jneilliii, 2019-09-24:
Apparently this may have already been fixed.  I just tried this on my OctoPrint instance and all tabs seem to work fine.

closed 2019-09-24 by jneilliii

StefanCohen, 2019-09-24:
Hmmm. I'm actually quit concerned about the webcam support. I have a feeling that there are quite a few conflicts between different plugins trying to hijack the stream from the control tab. I have seen what you reported myself but I can't reproduce it anymore. I've also had users experiencing conflicts with TouchUI while others says that its fine. At this point I'm actually leaning towards re-implementing the webcam settings in the plugin or possibly just remove webcam support altogether just to avoid this.

StefanCohen, 2019-09-24:
Another user just opened #48 so ther is something going on. Will have to look at this in the morning.

ejjenkins, 2019-09-24:
Just to add some clarity to this...the mention in the initial post above was my report on the octoprint github...I also opened #48. So this is the same one, not two different mentions of this. As a test, I just reinstalled Dashboard v1.5.0 and it still breaks my WebCam Tab.