j7126/OctoPrint-Dashboard #47: High bandwidth usage

TheLion, 2019-09-24:
I have very high bandwidth usage on my Windows computer, resulting in a hardly responding Octoprint webgui. This happens on multiple computers and with multiple browsers) on my Octoprint 1.3.11 running on Raspbian Buster on a RPi4 when opening the Dashboard Tab.

This strated happening a couple of days ago.

As long as I don't open the Dashboard tab, the issue does not occur), there is no issue.

I don't notice any high CPU (not higher that 10%) or other unusual activity.

I have an (almost) identical Octoprint running with another printer, same hardware, same versions of OctoPrint and Dashboard plugin, same plugins, and there the issue does not exist.

To illustrate:

[image]

On my other Octoprint:
[image]

I disabled the Dashboard plugin and everything is running smoothly again.

I tried to generate a debug log of the plugin, but none is generated, is that correct?

StefanCohen, 2019-09-24:
That is very interesting. I don't have access to a Win computer but I have compared the latest versions of Firefox to Safari and Chrome on OSX and I can't reproduce your results on my OctoPrint 1.3.11 running on OctoPi 0.16.0 on a Raspberry Pi 3 Model B Rev 1.2 nor on a Raspberry Pi 3 Model B Plus Rev 1.3

If you press f12 in firefox or Chrome and check the network tab, is there anything that stands out?

TheLion, 2019-09-25:
I'll check when I have a print running again. The load is higher than when a print is not running.

TheLion, 2019-09-25:
There is a lot of information in the Network tab. Nothing stands out (to me ;-) ) Anything specific you want to know ?

StefanCohen, 2019-09-25:
Is there anything more than the webcam that draws bandwidth? What framerate and resolution have you configured the webcam to stream? What is the CPU load on the PI during a print?

TheLion, 2019-09-25:
I actually turned the webcam off, to see if that changed anything. It didn't. So the webcam does not consume any bandwidth. The webcam is configured for 1280x720 at 10fps, but I guess hat doesn't matter, cause it is not showing on the dashboard tab. 

When I open Octoprint with the dashboard tab, the CPU load peaks to about 30%, which goes down after a couple of seconds to somewhere between 1 and 10. 

I also noticed that the browser, in my cause Firefox, kept consuming memory, up to 6Gb this evening when I started the print, up to the point that it hardly responded. When I closed the Octoprint tab, the bandwidth usage went down, but the memory usage did not, so I had to close Firefox to free it up. 

I'm still trying to wrap my head around the fact that my other Octoprint instance on another RPi, doesn't show the same issue. Network bandwidth usage is about half, even with the webcam showing. 

Both RPi's are connected by ethernet, not wifi.

StefanCohen, 2019-09-25:
This is very strange. I actually don’t know where to begin troubleshooting this.

TheLion, 2019-09-25:
I know. I know my way around troubleshooting networks, computers and software, it's my day-job ;-), but I am currently at a loss. 

Yes, I do have about 4. But they also run when using the other Octoprint. Just to be sure, I will disable them and do a test.

jwvanderbeck, 2019-09-26:
Noticing similar though not exactly the same issues.  Since I installed this plugin along the its dependency (the layer plugin) OctoPrint has become very laggy.  Before, switching to the OctoPrint tab, the page would show up nearly instantly, now it can take upwards of 30 seconds to a minute to show up.  Furthermore, while I don't see the high **network** usage as @TheLion , I am seeing very high **memory** usage from the OcotPrint tab now.  In excess of 2gb at times.

I have for now removed the dashboard but kept the layer plugin and want to go a few days like this to see if it improves or not.  If not, I will then remove the layer plugin.

Since I installed both at the same time I can't say for certainty which is the problem.

StefanCohen, 2019-09-26:
Ok. Thanks for the update. It is very hard to analyze this as I can't reproduce this. Do you see the same behaviour in other browsers too or is it only in Opera?

jwvanderbeck, 2019-09-26:
In my case it isn't even during a print.  I leave Ocotprint in a pinned tab in my desktop browser so basically it never gets closed.  I only print occasionally.

I can say that even though I only disabled the plugin last night I am already seeing a difference.  I can switch to the octoprint tab instantly again with no delay and mem usage is around 180mb according to Opera's task manager.

It is possible it is an interaction with some other plugin or plugins rather than this one on its own. Maybe something that modifies the UI is doing something that this plugin doesn't like?  I am using both Tab Order and Floating Nav Bar.

TheLion, 2019-09-26:
I am using neither Tab Order or Floating Nav Bar.

I do remember I played around with the Octo{rint-Stats plugin (https://github.com/amsbr/OctoPrint-Stats) a couple of days ago, which failed to install in the end. I can't find ant evidence of OctoPrint-Stats anywhere in my Octoprint. It is a very graphical plugin, so maybe that did mess something up.

StefanCohen, 2019-09-26:
I have no idea how to approach this at the moment. I can't reproduce it at all on any of my machines or RPis and I have spent a couple of hours on it allready. I have tried a huge number of plugins together with the Dashboard and I can't trigger the problem. I have seen a CPU usage of maximum 39% but that was with four separate and simultaneous webcam streams. The RPi has not gone over 20% CPU yet and the mem sits steadily at around 25% used. The bandwith used is just barely more than what is expected from the cam streams.

I can think of a few things to try to further pinpoint the problems you see:
Wifi issues? - Try wired networking?
Browser issues? - Try incognito mode?
Anything else you can think of?

I will however keep this issue open for the time being in case you find something or if anybody else reports the same thing.

TheLion, 2019-09-26:
I will try the incognito mode later today. Currently printing a 30 hour print ;-)
My Octoprint Pi's are already connected through ethernet and wifi is disabled.

jwvanderbeck, 2019-09-26:
Just to clarify, the memory usage I reported is memory used by the browser tab on my PC that has the Octoprint web page open, not memory usage of the RPi running octoprint.  I have no idea what its doing.  I'm not a linux guy.

jwvanderbeck, 2019-09-26:
Please correct me if I am wrong, but after some **very cursory** investigation, it appears to me that Dashboard is trying to display a webcam feed, even though it isn't the active tab.  Is this true?

Looking at the network diagnostics in Chromium, when the dashboard plugin is enabled I see very long network timeouts trying to enable a stream.  When it is disabled, that is not present in my network log.

Bikkies, 2019-09-27:
I'm seeing the dashboard plugin pull 60Mbit constantly, which is the same amount my control tab pulls with the plugin disabled, so it does seem to be that it just tries to keep the webcam stream active all the time even if you're not looking at it.

StefanCohen, 2019-09-27:
That is correct. I do not attempt to disable the stream if the tab is not in focus like the control tab does. In v1.5.1, the src for the stream is set to "" only if no webcam is configured in OctoPrint or if the webcam visibility in the Dashboard settings is turned off.

TheLion, 2019-09-27:
Incoginito mode in FF doesn't change anything.

I do notice that network bandwidth is about 20Mbps lower than before. Not running on v1.5.1 yet.

I'll keep monitoring.

StefanCohen, 2019-09-28:
Ooops. I didn't mean to close the issue. Any new findings?

jwvanderbeck, 2019-09-29:
1.5.1 seems better so far.  I've left it enabled since that release and the tab is still responsive.

StefanCohen, 2019-09-29:
Ok. Thanks for the update. I'm closing this issue now as there doesn't seem to be much I can do about this.

closed 2019-09-29 by StefanCohen