j7126/OctoPrint-Dashboard #48: OctoPrint-Dashboard Plugin breaks WebCam Tab Plugin

ejjenkins, 2019-09-24:
**Describe the bug**
When the release that added webcam view was installed it broke the WebCam Tab plugin.  When I installed your update the WebCam Tab no longer worked (blank tab).

**To Reproduce**
Steps to reproduce the behavior:
1. Install both plugins and Webcam Tab plugin no longer works.

**Expected behavior**
Both plugins to function.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Home Version 1903 build 18362.356
 - Browser FireFox 69.0.1 (64-bit) and Chrome Version 77.0.3865.90 (Official Build) (64-bit)
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jneilliii, 2019-09-25:
@ejjenkins, I've got both plugins installed and unable to reproduce the issue.  I don't have Octolapse though, have you tried with that plugin disabled yet? If you open your browser's developer console are there any errors displayed on initial connection/visit to the site or when you switch tabs?

ejjenkins, 2019-09-25:
I uninstalled OctoLapse and reinstalled Dashboard.  Still no change...WebCam Tab is still broken...the console shows:

Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_webcamtab:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#control:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_dashboard:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#control:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_webcamtab:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_dashboard:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#control:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_webcamtab:1 Unchecked runtime.lastError: The message port closed before a response was received.
/#tab_plugin_dashboard:1 Unchecked runtime.lastError: The message port closed before a response was received.

ejjenkins, 2019-09-25:
In the dashboard settings, if I enable webcam...it does not show.

ejjenkins, 2019-09-25:
The errors above only show in the Chrome console, not in FireFox...no errors show in FireFox, but it still doesn't work.

ejjenkins, 2019-09-25:
This is what @foosel posted on the Octoprint Github as well:

https://github.com/foosel/OctoPrint/issues/3281#issuecomment-534457885

ejjenkins, 2019-09-25:
I just installed v1.1.0 of Dashboard (prior to the webcam support) and everything works fine together...if I can help figurte this out please let me know.

ejjenkins, 2019-09-25:
v1.1.0 works fine
v1.2.0 works fine
v1.3.0 is where it breaks WebCam Tab

jneilliii, 2019-09-25:
I just realized one other thing that might be causing a difference in our installations.  I have my tabs rearranged in a way where the dashboard is not the first tab loaded when you open octoprint.  I'm curious if that would make a difference.  If you're willing to test that theory by installing either TabOrder or DragonOrder and put the temp tab as the first tab it would be helpful in knowing if that effects the outcome or not. Also, what is the setting you have for the webcam url?

ejjenkins, 2019-09-25:
Taborder is already installed...I'll reinstall the latest version of Dashboard and change the order around...

Stream URL is:  /webcam/?action=stream

jneilliii, 2019-09-25:
I can confirm, I am able to reproduce by putting the dashboard tab first. This leads me to believe it's related to the bindings and how things are loaded.  I'll work with @sco01 as best I can to resolve this.  I think the trick here is that if the webcam option is disabled in the settings it shouldn't break the webcamtab. Out of curiosity, did you cross-post an issue on the webcamtab plugin's repo?

ejjenkins, 2019-09-25:
OK, reinstalled Dashboard 1.5.0 and took Dashboard off of the first tab and the WebCam Tab is no longer blank, BUT the camera stream does not load...I get the Webcam stream not loaded error.

jneilliii, 2019-09-25:
I saw that a couple of times but it started working for some reason after a period of time.

ejjenkins, 2019-09-25:
I turned on the webcam on the dashboard tab and the webcam tab works...but only for a few seconds then I get the Webstream error

StefanCohen, 2019-09-25:
Good morning. I’ll try to reproduce this shortly.

jneilliii, 2019-09-25:
@sco01, not sure if this will help, but took a quick look at your code and I think you may need to add the onAllBound event handler to initialize stuff. I did it in my ConolidateTempControl plugin [here](https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/blob/5107b99bedc250e6bcb897c1f5c889fc0c59d454/octoprint_consolidate_temp_control/static/js/consolidate_temp_control.js#L46).

StefanCohen, 2019-09-25:
Ok. I’ll do that and will report back.

ejjenkins, 2019-09-25:
It is getting late for me...I will have to pick this up in the morning...just let me know what you want me to try and I will.

jneilliii, 2019-09-25:
Me too, I'm off to bed now.

ejjenkins, 2019-09-25:
I just saw the question if I posted this on the WebCam Tab plugin page...I did not.

jneilliii, 2019-09-25:
I think the issue is really tied to the webcamtab plugin in this situation.  When looking at it's code, it's overwriting the control tab's internal ontabchange function rather than augmenting it, which in turn is causing your calls to those functions to be broken because it's replacing the "selected tab" variables you're passing to the function.  Take a look over [here](https://github.com/malnvenshorn/OctoPrint-WebcamTab/blob/abe91bd4330b55b4ce6f00a918fa6199deb15ede/octoprint_webcamtab/static/js/webcamtab.js#L13).  You may be able to modify your code to detect if the webcamtab is loaded to modify those "selected tab" variables and use it's internal name rather than control.

StefanCohen, 2019-09-25:
Yup. I will investigate how to solve that but I think that leads to a bigger question that we probably need to bring up with Gina. These hacks to get around the control tab closing/starting the stream are starting to mount up. It is not a good sign that plugins needs to be aware of and work around each other in order to use the webcam stream.

foosel, 2019-09-25:
I agree and am open for suggestions. Though keep in mind that all my focus is currently on getting 1.4.0 out because of the py2.7 EOL situation.

StefanCohen, 2019-09-25:
I may have found something. I don't know why but It is probably related to the plugin loading order and/or the data binding order. I removed the __plugin_settings_overlay__ that put the dashboard first of the tabs and the Dashboard now works together with the WebCam Tab. I have also removed the entire controltab hack (self.onTabChange, self.controlViewModel.onBrowserTabVisibilityChange) and it still works. 

I have tested a few combinations of plugins after this fix and these are my results:

WebcamSB + ConsolidateTempControl + WebCamTab  + Dashboard = WebCamTab broken
WebcamSB + ConsolidateTempControl + WebCamTab = WebCamTab broken
WebcamSB + ConsolidateTempControl + Dashboard = All three ok
WebcamSB + WebCamTab + Dashboard = All three ok

So, The Dashboard now works with WebCamTab and ConsolidateTempControl as long as WebCamTab and ConsolidateTempControl are not enabled both at the same time. 
My conclusion is that this fix works but there is still a conflict going on between ConsolidateTempControl and WebCamTab. WebcamSB works no matter what and is not affected.

ejjenkins, 2019-09-25:
So the issue is (probably) with the method WebCamTab grabs the Stream.  Do you want me to try something on my system?

jneilliii, 2019-09-25:
I don't think it's necessary @ejjenkins.  We're able to reproduce the issue now that we know the combination of items. I'm going to apply the fix to my ConsolidateTempControl plugin when I get some spare time and @StefanCohen can choose to use the same fix or apply his own, which seems he may already have a potential fix.

jneilliii, 2019-09-25:
I have a working fix for the ConsolidateTempControl plugin now [here](https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/tree/0.1.3c).  Best I can tell it resolves the conflict between it and webcamtab, albeit that plugin steals my imported viewmodel's webcam image by design. I don't think that shouldn't effect you @StefanCohen because you're not importing the control tab's view model in the jinja file like that plugin does.

jneilliii, 2019-09-25:
I think there is still an issue with onBrowserTabVisibilityChange though. If you minimize, or switch browser tabs while on the webcam tab and then go back to that tab or restore the window the stream stops shortly after.

StefanCohen, 2019-09-25:
Thanks. I will try it in the morning. I’m will also try tab reordering a bit more to see if there are more ways to trigger webcamtab to fail.

jneilliii, 2019-09-25:
So I do remember the ordering was one of the reasons for the onAllBound event function that way if it's the first/default tab to load it would start the webcam.

StefanCohen, 2019-09-26:
I have tried your update and now ConsolidateTempControl works with WebCamTab and Dashboard (Meaning: all three displays the stream). I have however noticed that switching back and forth between ConsolidateTempControl and WebCamTab still causes WebCamTab to loose the stream so there is still something going on. The dashboard is however never affected of this behaviour as long as I don't attempt to put it first in the list of tabs.

jneilliii, 2019-09-26:
It's a timing thing. If you jump back and forth quickly it dies that. If you wait a bit on the tab and them change it's typically fine.

StefanCohen, 2019-09-26:
Ok. That makes sense.

I have unfortunatelly not managed to to get your example from ConsolidateTempControl to work with the dashboard. I'm currently leaning towards either removing the webcam completely from the dashboard or provide a user setting to bind it directly to the source stream on as a way to circumvent the problem.

jneilliii, 2019-09-26:
Can you share your dev branch so I can see the code your working with?

StefanCohen, 2019-09-26:
Hang on. This is probably much easier than I thought and the solution will be totally independent on other plugins. 

It makes total sense for the ConsolidateTempControl and WebCamTab to do what you did in your update since those plugins both tries to move exising objects, but not for the dashboard. To make it truly independent, I will simply create a new image object for the stream and implement my own enable() and disable() methods whenever the dashboard goes out of focus.

jneilliii, 2019-09-26:
That makes complete sense to me.  The sidebar plugin does exactly that (but probably not the enable/disable stuff) if I'm not mistaken and is why it worked no matter what the combination of installed plugins.

StefanCohen, 2019-09-26:
@ejjenkins It took a whil but I think I have a fix for your issue. I have tried it together with all plugins I can find that displays a webcam view and it seems to work regardless of tab order. I will bundle it up as v1.5.1 and release it later tonight.

closed 2019-09-26 by StefanCohen

ejjenkins, 2019-09-26:
I appreciate the effort on this!

Thank you very much.

ejjenkins, 2019-09-26:
I installed v1.5.1 and everything seems to be working as it should.  Well done and thank you!