j7126/OctoPrint-Dashboard #50: Fullscreen mode option when accessing the page

freddujour, 2019-09-28:
Hi there, very good plugin. smooth design
I am using a 22" screen to monitor my printer and would like to open it automatically in fullscreen mode when the page is accessed. Is there any way to request it via the URL?

Thanks

StefanCohen, 2019-09-28:
Not via the URL AFAIK. I have used the fullscreen API for the Dashboard. There is a javascript embedded in the plugin that opens the Dashboard container in fullscreen when the user clicks the icon. I'm not aware of any browser that allows that without manual intervention (klick, keypress, etc...) for security reasons.

j7126, 2019-09-29:
As already said by StefanCohen I don't believe it is possible to enter full screen without user interaction such as  a button press, however it may be possible to request via url that the dashboard takes up the entire window, so not true full screen but it gets us closer to what you are looking for. I will look into adding this. 

That in combination with a kiosk browser such as https://play.google.com/store/apps/details?id=de.ozerov.fully or https://www.microsoft.com/en-au/p/kiosk-browser/9ngb5s5xg2kp may be able to achieve what you are looking for.

j7126, 2019-09-29:
@freddujour what operating system and browser is being used for the screen?

freddujour, 2019-09-29:
thanks for advising. I am using both windows OS+ ie and raspbian+chromium on a second kiosk

j7126, 2019-10-01:
I have made the ability to access a full page mode by adding `?dashboard=full` to the end of the url.

Pull request #58

freddujour, 2019-10-01:
Super!

StefanCohen, 2019-10-02:
Fixed in v1.8.0

closed 2019-10-02 by StefanCohen