j7126/OctoPrint-Dashboard #53: 1.0.6 breaks M117 command!

ochm, 2019-09-29:
When updated Dashboard to version 1.6.0, printer stopped display about Octopi Connected,disconnected. 
After short investigation i foud that M117 is not sent, is not visible in serial monitor log.
Ony some like M117 x123 is vissible, but no M117 Hello world

**To Reproduce**
Try to send M117 Something and do verify what is displayed on printer.

**Expected behavior**
Display on printer what was sent by M117.

When the dashboard plugin is disabled od uninstalled, everything about M117 goes well.

StefanCohen, 2019-09-29:
Thanks. I think I know what went wrong and will try to fix that.

StefanCohen, 2019-09-29:
Confirmed. It was a botched GCODE parser introduced in v1.6.0. Fixed in v1.6.0.

closed 2019-09-29 by StefanCohen