j7126/OctoPrint-Dashboard #54: 1.6.0 can't send klipper Extended G-Code Commands

escapac, 2019-09-29:
I updated to 1.6.0 and now I can't send any klipper extended G-code commands to my printer.
Before update to 1.6 klipper extended G-code commands worked. I disabled dashboard plugin only and problem disappears.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to terminal or any other place where you can send g-codes to printer.
2. Send example restart command to printer or any other klipper extended G-code commands.

**Expected behavior**
With restart command klipper should restart.

 - OS: WIN 10
 - Browser Vivaldi
 - Version 2.7.1628.33 (Stable channel) (32-bit)

StefanCohen, 2019-09-29:
Thanks. This is probably the same as #53 but I will take a look just in case it something else causing this.

StefanCohen, 2019-09-29:
Confirmed. It was a botched GCODE parser introduced in v1.6.0. Fixed in v1.6.1.

closed 2019-09-29 by StefanCohen

StefanCohen, 2019-09-29:
I have now passed a full print on my Klipper printer.