j7126/OctoPrint-Dashboard #55: Axis over-run and limit issues after upgrading to 1.6.1

ruralaccent, 2019-09-29:
The Issue
I've just updated to 1.6.1 and something has messed up with my printer. Whenever I run a print via Octoprint, it now mashes the extruder into the limit switches and make an horrendous grinding noise - it doesn't do that with prints from the SD card. Disable the dashboard plugin, and it goes back to normal and everything works again. Re-enable the dashboard plugin, all hell breaks loose, disable it, back to normal. Whether it's the plugin itself or a newly-introduced conflict with another plugin, I know not, but it has only started with version 1.6.1 and I have made no other changes to the system at all, nor any other plugins. 

System
FF Creator Pro with the current stable version of Octoprint on a RP2B+, with Dashboard 1.6.1.

StefanCohen, 2019-09-29:
Can you please attach the gCode file so I can analyse what happens?
I would also like to know:
Slicer and version.
Printer Firmware and version. 

Overrunning limit switches seems very strange. Those limits should be enforced in the firmware (assuming the machine is homed prior to printing). Where do you define the homing procedure? Is it in the slicer or in Octoprint?
What happens if you home manually before printing via Octoprint with the Dashboard enabled?

ruralaccent, 2019-09-29:
Hi Stefan. I'll try to get you some more data tomorrow. I'll also try to get a video of the issue. It's definitely down to the plugin, but what specifically, I don't know. Looking at it again after I logged the issue, I'm not sure if it's an homing issue or limit issue (or both), but it makes a hell of a noise, nothing homes or prints. Ironically, the previous version worked fine for me! Many thanks, David 

David Baldwin 
BSc (Hons) MSc MRTPI
Chartered Town Planning Consultant 
david.baldwin@ruralaccent.com 

https://github.com/StefanCohen/OctoPrint-Dashboard/issues/55#issuecomment-536339315

ochm, 2019-09-30:
it is not about gcode, just type in terminal M117 message and there is nothing displayed on connected prinrer or catched in seial log.

ochm, 2019-09-30:
> 30. 9. 2019 v 8:12, Martin Och <martin@och.cz>:

wrong thread

ManuelW77, 2019-09-30:
Hey there,
i have the same problems with the latest release. gcodes like homing G28 are ignored, 
so limits are overrunned. Disabling the Dashboard Plugin brings back full functionality.
[abs.gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3669431/abs.gcode.zip)

StefanCohen, 2019-09-30:
Ok. I can't reproduce it on my physical printer so I need more input. Do you know if the G28 is coming from your slicer or from the Octoprint gCode Script?

ManuelW77, 2019-09-30:
It's coming from my Slicer. I user S3D and have all pre gcodes set in the slicer.
I also figured out, if the Plugin is enabled, i can't move the printer by writing gcodes in the Terminal tab. Plugin disabled, it works again.

ManuelW77, 2019-09-30:
And, this behaviour started with v1.6.0 of the Plugin.

StefanCohen, 2019-09-30:
Allright. And you have upgraded to 1.6.1?
Can you please send an octoprint.log, serial.log and a gcode file?

ManuelW77, 2019-09-30:
Hmm, i think so. Pluginmanager says no new updates. I'll check this if my running print is ready in round about 1h and give you a answer.

StefanCohen, 2019-09-30:
You should be able to see the version number in the plugin manager.

StefanCohen, 2019-09-30:
I just pushed 1.7.0 Can you please go thru the following steps:

1. In Octoprint Settings: Software Update -> Advanced. Klick "Force check for update (overrides cache used for update checks)" 
2. In the "Update Available" pupop, you should see "Dashboard Plugin: 1.7.0". Klick Update Now. and then restart.
3. After restart, go to Plugin Manager and locate the Dashboard. Verify that it says "Dashboard (1.7.0)"
4. Connect to your printer, go to the terminal and send "M117 Test". Verify that the printer LCD message is changed to "test.
5. In the terminal, send "G28". Verify that the printer is homing.

If 4 and 5 fails, please send the octoprint.log, serial.log and gcode file

ManuelW77, 2019-09-30:
My printer is ready now and i checked it. Installed Version is 1.6.0 and there are no updates available, even i try'd with force update. (Letzter Cacherefresh: 30.09.2019 11:23)

StefanCohen, 2019-09-30:
Ok. Then we at least know what the problem is. The bug you experience was introduced in 1.6.0 and fixed in 1.6.1. There must be a problem with the version propagation from Github to the Octoprint plugin manager. You can manually upgrade to 1.7.0 by first uninstalling 1.6.0, restart, then manually install in the plugin manager using this URL:  https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

StefanCohen, 2019-09-30:
@ruralaccent Can you please also verify the Dashboard version reported in the Plugin manager? Is it v1.6.0 or v1.6.1?

ManuelW77, 2019-09-30:
So, i can confirm, Pluginmanager said 1.6.0. 
But now the update to 1.7.0 comes up in the Pluginmanager and i installed it.
For now i can say, all commands in Terminal are working again. I'll do a test print later.

So thanks a lot for help and keep this nice Plugin working :)

StefanCohen, 2019-09-30:
No problem. I'll quite sure that 1.6.1 actually fixed the problem but I will keep this thread open until @ruralaccent can confirm.

ruralaccent, 2019-09-30:
Hi Stefan. I've just booted everything back up this morning and spent time looking into it all. I followed your steps to install 1.7.0 and that is working fine, with the issue resolved. I run three different instances of octoprint one three different printers. I'm embarrassed to say that I think I may have logged into another one which I had already updated to 1.6.1 when I went to check the plugin version before logging this bug report. When I checked the printer with the issues again this morning to provide you with the info required, it was showing 1.6.0, not the 1.6.1 that I had thought it was running. Humble apologies for any wasted time on this.

closed 2019-09-30 by ruralaccent