j7126/OctoPrint-Dashboard #57: raspberry temperature sensor (dht11)

iapetus696, 2019-09-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

StefanCohen, 2019-09-30:
Hi. The details of your request seems to have been lost.

iapetus696, 2019-09-30:
sorry ..... i have a temperature sensor (dht11) on my octoprint i wish i can see the temp in the dashboard like the CPU  temperature

iapetus696, 2019-09-30:
Now im using the enclosure plugin

StefanCohen, 2019-09-30:
Ok. I understand now. I agree that it would be a nice addition to the dashboard but I don't have that kind of sensor at hand so I don't feel comfortable supporting it when users asks questions about it.

iapetus696, 2019-10-01:
maybe a way to pull output values from other plugin like  enclosure plugin

StefanCohen, 2019-10-01:
Yes. I will take a look at that possibility.

StefanCohen, 2019-10-01:
I will reach out to the plugin developer and see if he is willing to add an event hook to get the temps.

iapetus696, 2019-10-01:
Thank you... i hope the best

StefanCohen, 2019-10-02:
No problem. My FR is available here: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/260

StefanCohen, 2019-10-02:
I went out and bought a dht11 today. It seems to work with the Enclosure plugin:

[image]

iapetus696, 2019-10-02:
yes work fine and with many sensors but i prefer all in the Dashboard

iapetus696, 2019-10-02:
all the values *

iapetus696, 2019-10-02:
Enclosure plugin is very good

StefanCohen, 2019-10-06:
I haven't received any feedback on the Enclosure plugin FR so I decided to give it a go myself this afternoon:

[image]

Is one sensor good enough for now?

iapetus696, 2019-10-06:
Thank you, one is good enough for me .... I think 2 is good idea .one for the  enclosure one for the fillament or the room temp .

StefanCohen, 2019-10-06:
The dash is getting crowded with data so one will have to do for now. This will be available in v1.10.0

closed 2019-10-06 by StefanCohen