j7126/OctoPrint-Dashboard #60: Extruded Filament widget not working

mdaneman, 2019-10-01:
**Describe the bug**
Not sure if this is a bug or I'm missing some external plugin, but the "extruded filament" display always shows "0.0m" for me.

**To Reproduce**
Steps to reproduce the behavior:
Just run a print and watch the filament display.

**Expected behavior**
Expect the filament display to show how much filament was extruded.

**Desktop (please complete the following information):**
 - OS: Windows10
 - Browser Chrome
 - Version 77

**Smartphone (please complete the following information):**
 - Device: iPhone6 / iPad
 - OS: iOS12 and iOS13
 - Browser Safari

StefanCohen, 2019-10-01:
Intersesting. What slicer are you using?

mdaneman, 2019-10-02:
I'm using Prusa Slicer latest version 2.1.0.  Here's a sample of g-code I just used.  The "filament used" indicator vacillated between "0.0m" and "-0.0m"

[03-socket-cradle._7.2mm_hole_0.2mm_PETG_MK3S_51m.gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3683135/03-socket-cradle._7.2mm_hole_0.2mm_PETG_MK3S_51m.gcode.zip)

StefanCohen, 2019-10-02:
Thanks. Seems like a good opportunity to play with Prusa Slicer :) 
I will take a look.

ejjenkins, 2019-10-02:
Mine is doing something similar...it bounces between 0.00 and -0.01 and 0.01.

I am using the latest version of Simplify3D (v4.1.2).
[Chip_Clip_-_Original_Sized_Version.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3683607/Chip_Clip_-_Original_Sized_Version.zip)

StefanCohen, 2019-10-02:
Thanks. I can reproduce it with that one too so I will investigate further. I'm not planning to buy S3D any time soon so I may ask for more gcode examples if I need them.

Naugrimohtar, 2019-10-03:
Note: I too am seeing this behavior (Prusa Slicer 2.1.0)

StefanCohen, 2019-10-03:
In the examples I have:

- S3D uses absolute positioning but keeps resetting the extruder during the print by setting G92 E0.0000
- Prusa Slicer uses relative extruder positioning.
- Cura uses absolute positioning

I need to find a way to support all three modes.

StefanCohen, 2019-10-04:
Update: I have made some progress but this is a lot trickier than I anticipated. The different slicers are handling this in a lot of strange ways making this Dashboard feature unreliable. I'll work on it some more but I'm currently leaning towards removing the feature.

StefanCohen, 2019-10-04:
Update. I found a way that works with all three methods (S3D, PS, Cura). I'm not yet sure that it covers all slicers but it will have to do for now.

ejjenkins, 2019-10-04:
Looks good!   Nice job.

StefanCohen, 2019-10-04:
Fixed in v1.9.0

mdaneman, 2019-10-07:
I wanted to report that now I do see filament widget update using both Prusa Slicer and Cura gcode.  Thanks for the fix.

tocache, 2019-10-28:
I'm having the same issue and I'm using the latest version 1.11.1, it shows me 0.0 of extruded filament:
[image]

But Octolapse shows this parameter correctly.
[image]

I'm using Prusa Slicer ver 2.1

[Sample_Spool_FIXED_0.2mm_PLA_MK3MMU2_6h47m.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3780933/Sample_Spool_FIXED_0.2mm_PLA_MK3MMU2_6h47m.zip)

StefanCohen, 2019-10-29:
@tocache I can't reproduce it using the gcode file you provided. The extuded filament is calculated and displayed correctly in my installs of 1.11.1:

[image]

Crouze, 2020-03-31:
Same problem here. Extruded filament shows 0.01m, sometimes jumping to 0.02m and back again. Octoprint 1.4.0, Python 2.7, Dashboard Plugin 1.11.4, PrusaSlicer 2.1.1.

StefanCohen, 2020-03-31:
Can you please provide a gcode file for me to reproduce it?

Crouze, 2020-03-31:
Sure. Here's one I'm printing right now.

[box3d_rear_cover_0.2mm_PLA_MK3S_6h5m.gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4409460/box3d_rear_cover_0.2mm_PLA_MK3S_6h5m.gcode.zip)

StefanCohen, 2020-03-31:
Very strange. I can't reproduce it using your gcode file. 

[image]

StefanCohen, 2020-03-31:
I will need to spend some time with this.

Crouze, 2020-03-31:
The counter does seem to increase, but very slowly...

[image]

StefanCohen, 2020-04-02:
There must be a setting somewhere in OctoPrint or another plugin that affects this behaviour but I can't find anything. The gcode file you provided just works for me regardless of what I do.

closed 2020-04-02 by StefanCohen

Autonomy02, 2020-09-02:
I am having the issue where the "Extruded Filament" remains at zero

Octoprint Environment:
Octoprint ver: 1.4.2
Python ver: 3.8.5
Octoprint running on Windows 10 Pro, direct connected via USB to the printer

Printer/slicer information
Printer: SnapMaker 2.0 A350
Slicer: SnapMaker Luban ver. 3.8.0

This is what is currently running:
[image]
[sewing_thread_cone_holder_3_gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/5163712/sewing_thread_cone_holder_3_gcode.zip)

Here is a smaller gcode file as well:
[sewing_thread_cone_holder_spacer_2_gcode.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/5163722/sewing_thread_cone_holder_spacer_2_gcode.zip)

larsbrockmann, 2021-12-27:
Did someone find the solution? I have the same problem. I am using SuperSlicer, a PrusaSlicer fork.

larsbrockmann, 2022-05-04:
@tocache Thanks for the response. Sadly, I can not remember if I ever solved the problem. I moved on to using Klipper and Fluidd.