j7126/OctoPrint-Dashboard #61: Any way to have it use less resources?

elvinhaak, 2019-10-01:
I have some problems with the plugin; it is somehow using up too many resources.
I have a couple of printers with Octoprint. One on a Pi3B (no other printers, just one) and one on a PiZeroW (all Raspberry). on the ZeroW it made the printer-process go very, very slow at always 100% processorusage so I needed to remove this plugin (and layer-guestumate) to be able to print well.

On the 3B the resources are used very much also but not slowing down, only making the pi react a bit slower but also heating up. At tooo high temperatures there is a fan but still the icon of throttle shows up after a longer print with fine details with these plugins enabled.

If you can select a response-time (like for checking only every second) it will most likely use much less resources. 
I do like the plugin though but cannot really use it at the moment because of this.

StefanCohen, 2019-10-02:
Hi and thanks for the feedback. This issue shares some similarities with #47 but is not identical. I haven't been able to reproduce the problem you experience. I can understand if it happens on the zero since it has very limited resources but the 3B should run without problems.

I have two RPis for development and testing: 

- Raspberry Pi 3 Model B Rev 1.2 
- Raspberry Pi 3 Model B Plus Rev 1.3

They are both fanless and doesn't even have a heatsink. None of them show the problem you experience.

With all the features enabled in the dashboard, the RPi CPU never goes above 20% CPU during a print and the CPU temp settles between 57.0°C and 63.0°C.

I am very curious about what can cause this difference. Can you please report what version of OctoPrint and OctoPi you are using. I would also like to know if you are using other plugins and in that case what versions of those.

The dashboard does very little processing on its own. Except for the GCODE processing to figure out the current extrusion, most of the heavy lifting is done by OctoPrint itself and the DisplayLayerProgress plugin. The exception to that is the psUtil processing of RPi system data that is done once 30 times/minute. I can try to lower that to every third second but I doubt that it will make any difference at all.

closed 2019-10-04 by StefanCohen

elvinhaak, 2019-10-04:
Hi, Did you receive my answers and info? I have replied it to your mail including pictures (screenshots).

StefanCohen, 2019-10-04:
Hi. No, I haven't received your answers.

elvinhaak, 2019-10-04:
Ok, I will send the mail again - it should be in your mailbox. 
The mail writes that I can respond directly so I did that with all the attachment about the different pi's and results.

By the way, also using FireFox, but only on Linux-systems and I had a couple of 'hangings' on my small laptop watching 24/7 the pi to remotely control it. also since I had installed this plugins so it might be related. Did not think about that before.

elvinhaak, 2019-10-04:
Hi,
On the PiZeroW I had no problems before the installation of this plugin and
the needed DisplayLayerProgress . I don't know how to see if that is using
the most resources or if that is using less resources if triggered less
often. In htop it is using a lot of processes if these plugins are running
somehow.
But also on the 3B.

On the PiZeroW the plugins are shown in files 10-14.
That is attached to a Sidewinder thus capable of pretty high speeds
although I normally don't even use the highest speeds on it. With only
these plugins the Pi is running fine, not heating up or getting up higher
then 40% when printing. Of course during updating (I do that manual every
few days) it goes up to 100% and shortly when uploading files. When I use
the Dasboard plugin (so also the DisplayLayerProgress plugin)  the
processor is going up on 100% almost all the time during printing.

On the Pi3b it is like shown in the rest of the screenshots. It is always
running a desktop for showing the display of octoprint so that I can see
what the temperatures are and how long the rest of the print will take. I
tried to put that on the dashboard-plugin but the screen is too small to
read that so I only display the temperature-screen and only showing the
Dashboard plugin on the PC-monitor. That one is connected to cabled network.
The printer on that is the Anet A8 plus.

StefanCohen, 2019-10-05:
Thanks. I still haven't received your email. It is probably better to upload the pictures here.

Regards,
Stefan

elvinhaak, 2019-10-05:
Strange. I do receive your mails when you answer me but somehow  you don't receive my mails. I try copy-paste...

elvinhaak, 2019-10-05:
Hi, 
On the PiZeroW I had no problems before the installation of this plugin and the needed DisplayLayerProgress . I don't know how to see if that is using the most resources or if that is using less resources if triggered less often. In htop it is using a lot of processes if these plugins are running somehow.
But also on the 3B.
On the PiZeroW the plugins are shown in files 10-14.That is attached to a Sidewinder thus capable of pretty high speeds although I normally don't even use the highest speeds on it. With only these plugins the Pi is running fine, not heating up or getting up higher then 40% when printing. Of course during updating (I do that manual every few days) it goes up to 100% and shortly when uploading files. When I use the Dasboard plugin (so also the DisplayLayerProgress plugin)  the processor is going up on 100% almost all the time during printing.

On the Pi3b it is like shown in the rest of the screenshots. It is always running a desktop for showing the display of octoprint so that I can see what the temperatures are and how long the rest of the print will take. I tried to put that on the dashboard-plugin but the screen is too small to read that so I only display the temperature-screen and only showing the Dashboard plugin on the PC-monitor. That one is connected to cabled network.The printer on that is the Anet A8 plus.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

elvinhaak, 2019-10-05:
Today with the newest version it seems better although the processor-usage is somehow growing as is the temperature of the Pi. Just in for a bit more then an hour now.

Very strange though that the hight is different on the same screen, don't know if it is related.
[image]