j7126/OctoPrint-Dashboard #64: Model Info

StefanCohen, 2019-10-03:
It would be nice to get a model info summary (modal dialog?) by clicking the gcode file name. The metadata is usually available as comments in the gcode file. 

Research:

- Is this already provided by another plugin or do we need to parse it from the file?
- How to determine slicer and version reliably?
- How to render it without cluttering the UI?

ManuGithubSteam, 2019-10-08:
Prusa slicer does provide this info direct in the gcode:

- used filament in m
- used filament In mm∆3
- total print time
- object dimensions ( I think)

I think a click or hover over the filename should give a info box with text. That's not too cluttery and you could turn it off In the settings.

PlusaSlicer does include "sliced with prusaslicer x.x version in the gcode

Hope this helps

StefanCohen, 2019-10-12:
I have decided to not implement this as there are existing plugins that can provide this info:

[SlicerSettingsParser]( https://plugins.octoprint.org/plugins/SlicerSettingsParser/)
and [SlicerSettingsTab](https://plugins.octoprint.org/plugins/SlicerSettingsTab/ )

closed 2019-10-12 by StefanCohen