j7126/OctoPrint-Dashboard #65: MultiCam support

PRGeno, 2019-10-03:
You have made so much progress in such a short time.  Great job on a real nice addition to OctoPrint.

I love that you added support for a webcam.  Would consider adding support for a second like what the MultiCam plugin has done?  I use MultiCam with front and rear view cameras to toggle between them to keep an eye how things are progressing.  It's just a second http address and a way to toggle between them.

Thanks for your consideration and thanks for Dashboard as well.

StefanCohen, 2019-10-03:
I do have a second webcam so I will keep this one for a rainy day but I have some things I want to do before this.

PRGeno, 2019-10-05:
That would be great if it could be done.  If I can add one more wish to this.  If possible please support the "Fullscreen Plugin" from Paul de Vries as well so double clicking on the webcam image will expand it to the full size of the browser, just like it would from the "Control" tab.

I'm enjoying your plug in and really appreciate you sharing it with the rest of us.

Kizan, 2020-01-29:
I would very much like to have multi-cam support. 

Thanks for this great plugin!

StefanCohen, 2020-03-08:
Update: I'm will not actively work on this feature but will happily review a PR if provided.

StefanCohen, 2020-04-02:
@PRGeno I looked into the possibility to support the FullScreen plugin today but there is unfortunately nothing i can do on my end to make that plugin work together with the dashboard. The reason is that FullScreen is hard-coded to look for a specific set of element IDs on the control tab that I can't hijack without breaking the control tab.

closed 2020-04-02 by StefanCohen

PRGeno, 2020-04-02:
@StefanCohen Thanks for giving it a shot.  Love your work and how great Dashboard has evolved.

StefanCohen, 2020-06-02:
This has been implemented in v1.13.0 thanks to a patch provided by the user CynanX.