j7126/OctoPrint-Dashboard #68: Linux ressources

overload08, 2019-10-04:
Hi !

Is it possible to add linux ressources (CPU, RAM, temperature) like Raspberry ?
All sensors are zero...

Thank's

StefanCohen, 2019-10-04:
it looks like psutil doesn't return anything.

What happens if you try the following from a terminal:
```
source ~/oprint/bin/activate
python
>>>import psutil
>>>psutil.virtual_memory().percent
```

overload08, 2019-10-04:
Hello !
It's look ok...

```
sysadmin@octo:~/OctoPrint$ source venv/bin/activate
(venv) sysadmin@octo:~/OctoPrint$ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> psutil.virtual_memory().percent
20.1
>>>
```

StefanCohen, 2019-10-04:
Ok. Then it should work as it is (maybe except for the cpu temp). I have no idea why it doesn't return anything for you but you can take a look at Issue #20 where a user had problems when running an older version of Octoprint. He solved it by upgrading. Can that be the issue for you?

overload08, 2019-10-04:
My octoprint is up to date :(
I'll try to debug...

StefanCohen, 2019-10-04:
Please do. It would take me some time to set up a linux container for debugging this.

You will find the relevant function (octoprint_dashboard) in __init__.py in the octoprint_dashboard folder.

overload08, 2019-10-04:
I think I see the issue.
As you can see, "cpu-thermal" object doesn't exist : 

```
>>> import psutil
>>> thermal = psutil.sensors_temperatures(fahrenheit=False)
>>> cpu_temp = round((thermal["cpu-thermal"][0][1]))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyError: 'cpu-thermal'
>>> print thermal;
{'acpitz': [shwtemp(label='', current=40.0, high=60.0, critical=60.0)], 'coretemp': [shwtemp(label='Core 0', current=57.0, high=84.0, critical=100.0), shwtemp(label='Core 1', current=55.0, high=84.0, critical=100.0)]}
>>> print str(psutil.cpu_percent(interval=None, percpu=False))
2.2
>>> print str(psutil.virtual_memory().percent)
20.5
>>> print str(psutil.disk_usage("/").percent)
5.7
>>>
```

May be include an exception and try other names for CPU sensors

StefanCohen, 2019-10-04:
That explains it. I had a hunch when i wrote "The CPU-temp will likely only work on a Raspberry Pi." in the Known Limitations section in the README.md and this confirms it. The problem is that I have no idea what these sensors can be named on other systems. But I agree that It should at least fail gracefully.

You should be able to see your sensors using: psutil.sensors_temperatures() What is the one for the CPU named in your case?

overload08, 2019-10-04:
CPU label is "coretemp".
Can you send me the result of "psutil.sensors_temperatures(fahrenheit=False)" on Pi ?
I think I can write somethink...

StefanCohen, 2019-10-04:
Sure:

`>>> psutil.sensors_temperatures(fahrenheit=False)
{'cpu-thermal': [shwtemp(label='', current=59.61, high=None, critical=None)]}
`

overload08, 2019-10-04:
Can you try this ?

`def psUtilGetStats(self):
    temp_average = 0
    temp_sum = 0
    thermal = psutil.sensors_temperatures(fahrenheit=False)
    if "cpu-thermal" in thermal:
        self.cpu_temp = round((thermal["cpu-thermal"][0][1]))
    elif 'coretemp' in thermal:
        for temp in range(0,len(thermal["coretemp"]),1):
            temp_sum = temp_sum+thermal["coretemp"][temp][1]
        self.cpu_temp = temp_sum / len(thermal["coretemp"])
    self.cpu_percent = str(psutil.cpu_percent(interval=None, percpu=False))
    self.virtual_memory_percent = str(psutil.virtual_memory().percent)
    self.disk_usage = str(psutil.disk_usage("/").percent)
`

StefanCohen, 2019-10-04:
Sure. 

`temp_average = 0` Is this a leftover from your testing?

StefanCohen, 2019-10-04:
Looks good. You can try it out by installing from this url since it isn't in a release yet: https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip

StefanCohen, 2019-10-04:
Fixed in v 1.9.0

closed 2019-10-04 by StefanCohen

overload08, 2019-10-05:
It's ok on my side with 1.9, thank's !