j7126/OctoPrint-Dashboard #69: Video disappears sometimes

mdaneman, 2019-10-04:
**Describe the bug**
I have the "Show Webcam" checked in settings and it sometimes show the video on the Dashboard, but sometimes it doesn't show it.  It sill allocates space for the video, but it's blank.  The video can still be seen the normal way in the Octoprint control tab.  Toggling "Show Webcame" on an off doesn't help.

**To Reproduce**
Not sure if this is exactly consistent, but for me the steps are to start a print and then wait for awhile or try to open Octoprint form a different computer or phone.  The video tends to show up the first time you connect to Octoprint from a particular browser, but then may disappear when you come back to the same browser window later.  Again, this may not be the consistent every time.

**Expected behavior**
I expected the video to show up in all locations at all times.

**Screenshots**
From Windows 7 Chrome:
[image]

From iPhone 6S:
[image]

**Desktop (please complete the following information):**
I've seen this both on Chrome through a Windows 7 desktop and on an iPhone through Safari (iOS13)

StefanCohen, 2019-10-04:
I have seen this too but only in Safari on OSX and I haven’t yet figured out why it happens.

overload08, 2019-10-04:
This append on control tab too, not related to the plugin I think

StefanCohen, 2019-10-04:
Then it is most likely not related to the Dashboard.

StefanCohen, 2019-10-04:
I am not sure at all that this will help but I added a url parameter to the stream to prevent browser caching. This is something I have intended to do for a while anyway.

StefanCohen, 2019-10-04:
Attempted fix in v1.9.0

closed 2019-10-04 by StefanCohen

mdaneman, 2019-10-04:
Thanks.  For me this definitely happens on both iOS and Chrome on Windows.  Also, the control tab is still showing video, so it's not related to any issue with video not streaming.  I'll update to the latest version once my current print completes (tomorrow) and let you know if I still see the problem or not.

StefanCohen, 2019-10-04:
Please do. I have no idea how to investigate this further at this time.

mdaneman, 2019-10-07:
Unfortunately this issue is not fixed yet.  I'm still seeing video disappear sometimes.

StefanCohen, 2019-10-07:
Ok. We need some way to reproduce this. I don’t see it on any of my systems. Do you have any idea how to trigger the problem or how to debug the issue on your system?