j7126/OctoPrint-Dashboard #70: Additional features I would like to see.

ejjenkins, 2019-10-04:
I would like to be able to click on the following widgets to adjust the settings:

Tool Temp (I tend to change temp while printing...I change filament a lot and need to adjust things)
Bed Temp 
Fan Speed (on/off and speed change)

New widget to adjust Feed Rate
new widget to adjust Flow Rate

StefanCohen, 2019-10-04:
Thanks for the feedback. 

This is similar to #4

I have to say that I'm actually still sceptical to adding controls to the dashboard. The intention was never to replace the controls and temperature tabs and I don't want to clutter the UI with widgets for everything.  

I have however played a bit with controls as context menus in an attempt to avoid clutter and a few controls are already implemented (connect/disconnect, Pause/resume/Cancel). I tried implementing temp controls too but failed and have rolled it back for now. I may give it another try in the future but It has not my highest priority right now.

PRGeno, 2019-10-07:
To offer another user's perspective: I actually like NOT having this ability.  I prefer to keep the Dashboard a read only tab to prevent a wayward click affecting things in the middle of a print job.  If I want to change something I will consciously click over to the Control tab to do it, and then click back to the Dashboard tab for monitoring.  Just offering another use case perspective on this topic.

StefanCohen, 2019-10-07:
Good feedback. Thanks.

ManuGithubSteam, 2019-10-08:
I second this please leave it as read only ;)

StefanCohen, 2019-10-09:
I have now decided to remove the existing context controls as they are too obscure to be useful. I will instead keep the Dashboard mostly read-only for the time being. The exception is #87 for fullscreen users.

closed 2019-10-09 by StefanCohen