j7126/OctoPrint-Dashboard #71: Full screen mode is bigger than screen

ochm, 2019-10-04:
**Describe the bug**
In full screen of dashboard i see on my both monitors the most low informations out of screen. Both are digitaly connected, there is no issue with screen size settingand i see no issues in other applications or windows main bar. 

**To Reproduce**
Start fullscreen display

**Expected behavior**
Is it possible to move whole content of fullscreen approx. 0.5cm higher?

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
W8.1
resolution 1920x1080

Thank you.

StefanCohen, 2019-10-04:
I have a similar problem in Safari. What browser are you using?

Tagging @j7126

ochm, 2019-10-04:
Currently I'm using Chrome, in the FF ESR is the same problem.

j7126, 2019-10-05:
I'll look into it.

LMS0815, 2019-10-05:
Same issue with Chrome on 1280x1024 Monitor

elvinhaak, 2019-10-05:
Kind of the Same 'issue'. Using  the Pi's display, using epiphany-browser.
The icons about the info are coming on screen but not any info. I'll try to make a screenshot using VNC. Actually, it is divided over multiple pages, but none with info although scrolling through them.

StefanCohen, 2019-10-06:
@elvinhaak That looks like a completely different Issue. Do you mind removing this and opening up a new one to keep this thread on topic?

elvinhaak, 2019-10-06:
ok I will

StefanCohen, 2019-10-11:
WIll be fixed in 1.11.0 I have reduced the bottom padding for the grid containers to free up some space. For fitting the dashboard on smaller screens, either reduce the number of widgets or use the bar progress type to save some vertical space.

StefanCohen, 2019-10-13:
Fixed in 1.11.0

closed 2019-10-13 by StefanCohen