j7126/OctoPrint-Dashboard #73: Ability to scroll Layer Duration

LMS0815, 2019-10-05:
If possible ad the ability to scroll the Layer Duration graph for big jobs...
[image]

StefanCohen, 2019-10-05:
I at least need to find a better algorithm for reducing the number of X labels :)

I use [chartist.js](https://gionkunz.github.io/chartist-js/) for the graph. It is possible that scrolling & zooming is supported but I haven't looked yet.

StefanCohen, 2019-10-06:
I just checked and chartist.js doesn't provide a method to scroll the chart. I'll leave this one open until I find a better way to reduce the X axis labels.

StefanCohen, 2019-10-08:
The current algorithm to remove labels based on the number of labels is defined [here](https://github.com/StefanCohen/OctoPrint-Dashboard/blob/e601b9edd5b0663170be5f2ae9405ce716d7c0ab/octoprint_dashboard/static/js/dashboard.js#L382-L389)

We'll need to find a method to keep it at a maximum of 40 labels but I can't figure out the math to do so. I welcome any suggestions.

Willmac16, 2020-06-06:
I know of a pretty simple way to allow for scrolling: increasing the width of the graph based off of the number of layers and correctly setting the overflow attributes for the div the graph lives in. This would require changes to the label interpolation algorithm to ensure even spacing, and it might make sense to add an option to enable/disable layer view scrolling.

Willmac16, 2020-06-06:
I also wrote label interpolation that does equal spacing maxing out at 40 labels.

StefanCohen, 2020-06-08:
Cool. I will review the PR. Thanks!

StefanCohen, 2020-06-09:
That PR looks really good. I'm including this in v1.14

Haasje, 2020-06-11:
Is there a way to disable the scrollbar? I think it looks really ugly in the theme I'm using and I just want to have a quick overview of the layertimes without having to scroll thru them.

StefanCohen, 2020-06-11:
Not yet. I still haven't figured out how to approach the settings for this.

Willmac16, 2020-06-11:
Does it make more sense to implement a plugin wide setting for this or just a simple toggle button in the dashboard?

ranger81, 2020-08-04:
Hi there,
I have another suggestion for the layer duration graph. Is there a possibility to include an "auto-scroll" to the most right position? For longer prints, the layer duration graph only shows the beginning of the print. Then you need to manually scroll to the right most end to see the lastest durations. I think it would be beneficial to automatically scroll to the latest durations.
What do you think?

Willmac16, 2020-08-14:
Do you think it would be better to have an option that only displays the last 80 or so layers instead.

Willmac16, 2020-09-30:
A layer duration graph scrolling option has been added.

closed 2020-09-30 by Willmac16