j7126/OctoPrint-Dashboard #74: octoprint  multi instances with muti printers , diferent color in fullscreen

manueloya, 2019-10-06:
hi, i have 3 octoprint instances and the fullsceen color is the same in the 3 where is the reference to the color? i'd like each full screen with the color of each printer.
thanks a lot for your job

StefanCohen, 2019-10-06:
This is the CSS selector for the div used in fullscreen:

```
.dasboardContainer.dashboard-full {
  color: #dadadc!important;
  font-weight: 400;
  background-color: #000000;
  user-select: none;
  width: 100vw;
  height: 100vw;
  position: fixed;
  top: 0px;
  left: 0px;
}
```
There are a few more available that are selected depending on the browser you are using.

There is however a script in dashboard.js that will set the current background colour based on what themeify theme you have selected.

@j7126 knows much more than I do when it comes to that CSS sorcery ;)

manueloya, 2019-10-06:
with iexprorer the icons are not sorted and the full screen button does not work, with the edge the icons are sorted and the full screen button works, but always catches the color of the first instance of the octoprint, and I have 3 runing . and with chrome, the icons are sorted but the full screen always appears without background color

StefanCohen, 2019-10-06:
Ok. I doubt that we can get IE to play nice but with chrome it should at least be theoretically possible to do what you want.

Have you seen this setting?
[image]

This makes the dashboard inherit the background colour of the current Octoprint theme (if available) when in fullscreen mode. The default is otherwise black in fullscreen mode.

I don't know much more than that and I don't see how it would be possible to set the CSS differently across instances (other than editing the CSS file on disk) but maybe there is an existing method for this?

j7126, 2019-10-06:
Just set the use theme colors in full screen mode option and then use a different theme with Themeify on each instance of octoprint.

StefanCohen, 2019-10-07:
@manueloya This should be working as @j7126 described. I’m closing the question now but feel free to re-open if you run into any issues.

closed 2019-10-07 by StefanCohen