j7126/OctoPrint-Dashboard #75: Some information is not displayed

ttraxx, 2019-10-06:
The printer does not run on the screen, but the information is not visible at the time of printing. Only those that are now visible are displayed. Where could the problem be?

Pi-W Prusa MK3S+MMU2S 

[image]

StefanCohen, 2019-10-06:
That looks like a data binding issue in the template. More specifically, it seems to be with the Fan Speed gauge. That gauge is fed with data from the DisplayLayerProgress plugin. You should have at least one Javascript error in the browser console due to this. Is that correct?

That Leads me to the batch of questions:. 

- Is the DisplayLayerProgress plugin installed and working?
- What version of Octoprint are you on?
- What version of OctoPi are you on?

StefanCohen, 2019-10-06:
Note to self: Should we move the plugin notification code to first in the displayviewmodel?

solyxius, 2019-10-07:
Something amiss on a RPi also ... 

[image]

[image]

[image]

StefanCohen, 2019-10-07:
@solyxius That doesn't seem to be related.

The dashboardViewModel (missing: gcodeViewModel) error message indicates that the gcode viewer tab is missing or modified in your system. Do you have any plugin installed that have altered that tab in any way?

solyxius, 2019-10-07:
@StefanCohen I don't believe so, only plugins installed are -

[image]

ttraxx, 2019-10-07:
> Vypadá to jako problém s vazbou dat v šabloně. Konkrétně se zdá, že je to s měřičem rychlosti ventilátoru. Tento měřič je napájen daty z pluginu DisplayLayerProgress. Z tohoto důvodu byste měli mít v konzole prohlížeče alespoň jednu chybu Javascript. Je to správně?
> 
> To mě vede k celé řadě otázek:
> 
> * Je nainstalovaný a funkční plugin DisplayLayerProgress?
> * Na jaké verzi Octoprint máte?
> * Na jaké verzi OctoPi jste?

plugin is now installed
Octoprint I have the latest
it's RPI-W (Prusa Upgrade)
The problem remains

solyxius, 2019-10-07:
@StefanCohen what's the method to roll back to a previous version (say 1.9.0) to troubleshoot if my issue is isolated to later versions or is inherent to a change in my install?

StefanCohen, 2019-10-07:
@solyxius I think this is a totally separate issue from the one described by @ttraxx Can you please open a separate issue for it?

StefanCohen, 2019-10-07:
@ttraxx RPI-W (Prusa Upgrade) It is the first time I’ve heard about that. Is it running on a PI Zero?

ttraxx, 2019-10-07:
@StefanCohen yes Raspberry Pi Zero W

ttraxx, 2019-10-07:
@StefanCohen prusa has instructions on how to integrate RPI primo directly into the printer. Therefore, this RPI model

StefanCohen, 2019-10-07:
Ok. I will take a look but i don’t have a zero w. I also doubt that the Dashboard will run on it since it is very underpowered to begin with and Gina strongly advice against running Octoprint on it.

ttraxx, 2019-10-07:
@StefanCohen long time booting, but runs correctly. I've been using it like this for months ... I'll try to eliminate the problem by temporarily using 3B +, but I don't believe it will affect the data display.

ttraxx, 2019-10-07:
@StefanCohen Should I add some data from the existing deployment?

StefanCohen, 2019-10-07:
@ttraxx can you confirm that Displaylayerprogress works? Please enable the toolbar and make sure that the layer info, height, etc is displayed correctly in the toolbar during a print.

StefanCohen, 2019-10-08:
@ttraxx Are you by any chance using a modified version of OctoPrint called [PrusaPrint](https://manual.prusa3d.com/Guide/0.+Preparing+necessary+components/585?lang=en#s9670)?

Please also see my previous question.

ttraxx, 2019-10-08:
[image]

ttraxx, 2019-10-08:
@StefanCohen 
hello
No i have official octoprint
Now i update dashboard plugin  to actual version
The problem remains. I attached a photo when printing with a functional plugin to display the height

StefanCohen, 2019-10-08:
Strange. If you open the browser terminal do you see any error messages?

ttraxx, 2019-10-08:
[Zobrazí/skryje detaily zprávy.] ReferenceError: $ is not defined[Zjistit více] 192.168.1.75:6666:5
Na nezabezpečené stránce (http://) se nachází pole pro zadání hesla. Jedná se o bezpečnostní riziko, v rámci kterého může dojít k odcizení přihlašovacích údajů.[Zjistit více] 192.168.1.75
Starting dependency resolution... packed_core.js:15314:9
... dependency resolution done packed_core.js:15424:9
Initial application setup done, connecting to server... packed_core.js:15728:9
Connected to the server packed_core.js:13464:9
Finalizing application startup packed_core.js:15706:13
Going to bind 35 view models... packed_core.js:15590:13
Could not bind view model TemperatureViewModel to target #temp : visible@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338 line 11374 > Function:3:60
update@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11409:229
m/</<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11378:286
Pc@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11357:355
Qc@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11357:128
aa@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11356:477
a.B@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11355:37
m/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11378:271
q@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11317:244
m@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11377:314
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:387
a.Rb@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11381:413
bindViewModels/</<@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15654:33
Pn@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11208:530
ur/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11229:66
bindViewModels/<@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15624:25
Pn@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11208:530
ur/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11229:66
bindViewModels@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15591:13
packed_core.js:15661:33
Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist packed_core.js:15643:33
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist packed_core.js:15643:33
Could not bind view model DashboardViewModel to target #tab_plugin_dashboard : attr@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338 line 11374 > Function:3:65
update@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11392:296
m/</<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11378:286
Pc@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11357:355
Qc@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11357:128
aa@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11356:477
a.B@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11355:37
m/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11378:271
q@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11317:244
m@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11377:314
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:387
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
g@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:278
k@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11375:440
a.Rb@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11381:413
bindViewModels/</<@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15654:33
Pn@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11208:530
ur/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11229:66
bindViewModels/<@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15624:25
Pn@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11208:530
ur/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:11229:66
bindViewModels@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:15591:13
packed_core.js:15661:33
User admin logged in packed_core.js:2443:25
... binding done packed_core.js:15682:13
Application startup complete packed_core.js:15702:13
[Zobrazí/skryje detaily zprávy.] TypeError: layer.filament[i] is undefined[Zjistit více] packed_core.js:8206:70
Error calling fromCurrentData on view model GcodeViewModel : GcodeViewModel/self._onLayerSelected@http://192.168.1.75/static/webassets/packed_core.js?b2b56004:8206:70
DashboardViewModel/self.gcodeViewModel._onLayerSelected@http://192.168.1.75/static/webassets/packed_plugins.js?aa2ceec4:1066:28
[... cut ...]
[5]</EventTarget.prototype.dispatchEvent@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:22528:5
[14]</</SockJS.prototype._transportMessage/<@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:23241:11
[14]</</SockJS.prototype._transportMessage@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:23239:9
[3]</EventEmitter.prototype.emit@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:22444:5
WebSocketTransport/this.ws.onmessage@http://192.168.1.75/static/webassets/packed_libs.js?2d66a338:25315:5
packed_core.js:14945:13

ttraxx, 2019-10-08:
@StefanCohen I have now used firefox instead of chrome to exclude the browser. Information of the ventilator is unavailable

StefanCohen, 2019-10-09:
> Could not bind view model TemperatureViewModel to target #temp : 

This can be the root cause of this problem. There seems to be a problem with your TemperatureViewModel binding. This happens before the Dashboard tries to bind its own viewModel (that is depending on the TemperatureViewModel) so it also fails. 

Do you see the same error if you uninstall the Dashboard?

closed 2019-10-11 by StefanCohen