j7126/OctoPrint-Dashboard #76: Dashboard height indicator incorrect

RyanParle, 2019-10-06:
**Describe the bug**

Current height and total height values reported incorrectly in dashboard plugin. As an example during a recent print I found that the values reported by the DisplayZ and DisplayLayerProgress plugins were reporting the current height as 123.79mm and a total height of 155.85 however the dashboard plugin is showing a current height of 124.69 and a total height of 154.85

**To Reproduce**

simply compare values between dashboard and DisplayLayerProgress

**Expected behavior**

The values to match

**Screenshots**

[image]

**Desktop (please complete the following information):**

Issue independent of OS / Browser

StefanCohen, 2019-10-06:
Thanks. Can you tell me what setting you have in the following dialog in the DisplayLayerProgress plugin settings:

[image]

StefanCohen, 2019-10-07:
Can you please also attach a gcode file that has this problem?

StefanCohen, 2019-10-08:
I can see this problem too with the latest DisplayLayerProgress release. This may be a problem with the data sent from DisplayLayerProgress thru the event handler. This is probably related to #81

StefanCohen, 2019-10-08:
Fixed in v1.10.2

closed 2019-10-08 by StefanCohen

RyanParle, 2019-10-08:
Hi Stefan, Thanks for the quick work on getting an update out, sorry i was not able to send you more information, I've been a little busy over the last few days.