j7126/OctoPrint-Dashboard #77: Broken update on PC/x86

Loloke, 2019-10-06:
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

StefanCohen, 2019-10-06:
That is caused by the DHT library needed for the sensor input added in v1.10.0. 
I need to check if it is possible to make it conditional in the installer somehow. Either that or three other options:

- Force RPi users to install it manually.
- Stop supporting non-RPi installations.
- Remove the feature.

j7126, 2019-10-06:
I have a fix for this

j7126, 2019-10-06:
fixed PR #78

solyxius, 2019-10-07:
Something amiss on a RPi also ...
[image]
[image]

StefanCohen, 2019-10-07:
Modified commit by @j7126 and added check to see if the setup is running on an RPi by trying to load the RPI.GPIO module. I have verified on both an RPi and in an Ubuntu LXC container and both worked. Fixed in v1.10.1 Please report back with a verification if it works or not.

overload08, 2019-10-07:
Hi !
Doesn't work on PC :

```
Oct  7 10:08:32 octo octoprint[16819]: 2019-10-07 10:08:32,792 - octoprint.plugin.core - ERROR - Error loading plugin dashboard
Oct  7 10:08:32 octo octoprint[16819]: Traceback (most recent call last):
Oct  7 10:08:32 octo octoprint[16819]:   File "/home/sysadmin/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.11-py2.7.egg/octoprint/plugin/core.py", line 863, in _import_plugin
Oct  7 10:08:32 octo octoprint[16819]:     instance = imp.load_module(module_name, f, filename, description)
Oct  7 10:08:32 octo octoprint[16819]:   File "/home/sysadmin/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 17, in <module>
Oct  7 10:08:32 octo octoprint[16819]:     import Adafruit_DHT
Oct  7 10:08:32 octo octoprint[16819]: ImportError: No module named Adafruit_DHT
```

StefanCohen, 2019-10-07:
@overload08 can you confirm that you tried with v1.10.1 and not v1.10.0?

overload08, 2019-10-07:
Yeah, I installed Adafruit_DHT and it was ok

StefanCohen, 2019-10-07:
Allright. I need to find a Windows machine and try it myself.

j7126, 2019-10-07:
I tested my PR on Windows and on a pi and it worked fine on both.

overload08, 2019-10-07:
Just a precision, i've tested on linux...

j7126, 2019-10-07:
I have just tested v1.10.1 on windows and it is working fine with no problems, it seems that this will require some further investigation.

j7126, 2019-10-07:
@StefanCohen 

The following screenshot is from an octoprint instance running on windows: 
[image]

And the following screenshot if form an octoprint instance running on octopi on raspberry pi:
[image]

It is not properly detecting the new version of the plugin that is available. When the update button is pressed it simply does nothing. I had to uninstall and re-install the plugin to get it to work. 
I am not yet sure if it is related to this but it is interesting.

solyxius, 2019-10-07:
On RPi I had to go to Advanced Options in the software updates and bypass cache to retrieve the latest version version

Loloke, 2019-10-07:
v1.10.1 works fine, thanks!

StefanCohen, 2019-10-07:
@j7126 The cache thing is a mystery. Have you tried what @solyxius describes? That usually works for me when this happens to me. I have noticed that it only happens for me when going from a .0 to a .1. 
Anyway. This seems to be working now. Thanks for the help!

closed 2019-10-07 by StefanCohen