j7126/OctoPrint-Dashboard #79: Broken after Update

LMS0815, 2019-10-07:
Not working any longer. 

[Error Screen](https://user-images.githubusercontent.com/8235214/66308680-f79fd200-e907-11e9-890b-614864b70d88.png)

Please DO NOT add all requested features.
Although I like _dashboard_, its getting overloaded.

StefanCohen, 2019-10-07:
Same problem as with #77 that was fixed in 1.10.1?

solyxius, 2019-10-07:
Same problem with v1.10.1 on a RPi also ... 

[image]

StefanCohen, 2019-10-07:
Interesting. So this is probably unrelated to the fix in 1.10.1 then. Any error messages in octoprint.log or in the browser console?

solyxius, 2019-10-07:
After rolling back to 1.9.0 ... therefore seems to be something to do with 1.10.x?

[image]

StefanCohen, 2019-10-08:
Hmm. Nothing strange with the versions. What about the browser console log? Any errors in there? 

I'm starting to wonder if this may be related to the introduction of the layer progress gauge in 1.10.0 that binds to the gcode tabs view model. It could be conflict another plugin that somehow disturbs that viewmodel.

Do you still see the problem if you temporarilly disable all other plugins except DisplayLayerProgress, restart Octoprint and reload the browser?

If the problem goes away then, can you please re-enable the plugins again and post the plugin loading section in your octoprint.log from right after the startup. It is the lines following `octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:`

That would allow me to install all your plugins locally and see if that can reproduce the problem.

solyxius, 2019-10-08:
@StefanCohen no joy ... all plugins disabled except -

[image]

[image]

However, there are browser errors related to gcode view model -

[image]

StefanCohen, 2019-10-08:
Thanks. It looks like the gcodeViewModel is missing when the Dashboard ViewModel is created. Do you agree @j7126 ?

StefanCohen, 2019-10-08:
Have you possibly disabled the Gcode Visualizer?
[image]

Disabling it causes the same issue on my instance:
[image]

The v1.10.0 released introduced a dependency to that tab for the new layer progress gauge.

LMS0815, 2019-10-08:
I never enabled _Gcode Visualizer_ it never was (because it is stuck anyway if the print job is big).
Is there any change to get _Dashboard_ again running without this dependency?

StefanCohen, 2019-10-08:
Ok. Now we know the reason.

 It should be enabled by default (it has at least been in all the versions of OctoPi and OctoPrint I have used so far) so it seems like this is something that has to be manually disabled by users.

I will play a bit to see what happens if we make it optional but I fear that it will break the Dashboard completely.

LMS0815, 2019-10-08:
Just for clarification...
**Dashboard** now depends on
_Gcode Visualizer_ and _Display Layer Progress Plugin_? 
Both needed to create Layer progress gauge?

LMS0815, 2019-10-08:
Yes I manually disabled it because it once crashed and never happend to work again.
Most of my prints are to big for _Gcode Visualizer_  (see description).

LMS0815, 2019-10-08:
> I will play a bit to see what happens if we make it optional but I fear that it will break the Dashboard completely.

Older versions have been working without  _Gcode Visualizer_

StefanCohen, 2019-10-08:
Correct but they are used for different purposes. Gcode Visualizer is used for the layer progress gauge. DisplayLayerProgress is used for layer height, model height, Layer Duration Graph, etc...

There are actually plenty of more dependencies. The complete list of ViewModel dependencies is:

["temperatureViewModel", "printerStateViewModel", "printerProfilesViewModel", "connectionViewModel", "settingsViewModel", "displaylayerprogressViewModel", "controlViewModel", "gcodeViewModel"]

StefanCohen, 2019-10-08:
> Older versions have been working without Gcode Visualizer
Yes. You can always use 1.9 or earlier. These versions didn't have that dependency.

LMS0815, 2019-10-08:
Got it. THX!
I prefer always to us the current version...  (Or not using it, if it is or suitable).
As OctoPrint does not have a Prefered Version switch for plugin updates, the next update would install the current version again.

StefanCohen, 2019-10-08:
No problem. I just tried making the loading of the viewModel optional but that didn't work at all. I will make sure that this requirement is properly described in the readme, releas notes and in the wiki. 

Sorry for the inconvenience. As I said, I wasn't even aware that it is possible to disable the gcode viewer.

@solyxius Does enabling the feature solve your problem?

LMS0815, 2019-10-08:
I can not get gCode viewer to work. - Others might have the same issue...
[image]

StefanCohen, 2019-10-08:
I'm going to investigate this a bit more to see if we can fail more gracefully but I will close this Issue now that we know the reason and what to expect. 

I will however open a new Feature Request and see what we can do in a future version. The dependency is required for the Layer Progress widget, a feature that was planned from day one. I feel that It may be possible to work around the error and disable the feature for situations like yours. Feel free to provide a PR if you want to help.

closed 2019-10-08 by StefanCohen

solyxius, 2019-10-08:
@StefanCohen re-enabling Gcode Visualizer solved the issue, thanks!