j7126/OctoPrint-Dashboard #80: Dashboard loads webcam stream upon loading and never unloads it

EpicLPer, 2019-10-07:
**Describe the bug**
I was wondering why my PC constantly downloaded ~3MB/s so I checked the Ressource Manager and found the OctoPi tab as the issue. Once I disabled the Dashboard Plugin the issue instantly went away.

**To Reproduce**
Steps to reproduce the behavior:
1. As soon as Dashboard is installed it'll load the webcam stream
2. If this doesn't work enable, save, then disable the "Show Webcam" in the settings. Bug will now occur.

**Expected behavior**
Webcam Stream shouldn't be loaded upon finishing loading the page or should be unloaded if tab isn't active for a few seconds like the normal Control page does it.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - Device: Desktop PC
 - OS: Windows 10 1903
 - Browser: Chrome, 77.0.3865.90
 - Version: 1.10.1

StefanCohen, 2019-10-07:
That is correct. I have not implemented this yet.

StefanCohen, 2019-10-12:
Partly fixed in 1.11.0

The webcam stream is not loaded if "Show Webcam" is disabled. I will not implement the loading/unloading of the stream on tab change as I couldn't get that to work reliably.

StefanCohen, 2019-10-13:
Fixed in 1.11.0

closed 2019-10-13 by StefanCohen