j7126/OctoPrint-Dashboard #81: Layer count not correct in dashboard but in DisplayLayerProgress

ManuGithubSteam, 2019-10-07:
**Describe the bug**
The layer count on the octoprint and in Display Layer Progress semms to nearly correct. see the Left side of the picture.
However in the dashboard some other values are shown....

**To Reproduce**

Print the gcode.

**Expected behavior**
Number schould be the same as Display Layer Progress.

**Screenshots**
See the numbers in the screenshot:
[image]

**Desktop (please complete the following information):**
 - OS: Raspian
 - Browser Chrome
 - Version 1.10.10

StefanCohen, 2019-10-07:
Hi! The value above the cursor in the screenshot is not coming from displaylayerprogress (and I doubt the model is 1m tall but I might wrong) please enable the displaylayerprogress toolbar and compare with the dashboard values. Are they the same?

ManuGithubSteam, 2019-10-07:
Here is a full screenshot with the output from DisplayLayer Progress in the statusbar on top left side.
They are the same as on the octoprint Sidebar.
The first layer count there is from octoprint itself and it shows the same values.

[image]

So we have 3 places with layer 98 one of them is octoprint itself -> Dash does some wrong math i belive?

StefanCohen, 2019-10-07:
Very interesting. So the model is really 1000mm tall? I'm not doing any calculations at all but simply display what displaylayerprogress reports via the event bus. Can you please upload the Gcode so i can debug this?

ManuGithubSteam, 2019-10-07:
Sure thing there you go.
No the model is 10cm tall.
[OnePiece_Spool_Support_3h7m_Real PETG - 0.25mm QUALITY - MK3S - Volcano 0.4.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3699321/OnePiece_Spool_Support_3h7m_Real.PETG.-.0.25mm.QUALITY.-.MK3S.-.Volcano.0.4.zip)

StefanCohen, 2019-10-07:
This may be the same issue as#76

ManuGithubSteam, 2019-10-07:
Im not sure, the layer count in #76 is correct. Here it is not.

StefanCohen, 2019-10-07:
Thanks. I will take a look tomorrow. 100mm seems more likelly compared to the 1000mm in the screenshot. What version of displaylayerprogress are you using?

ManuGithubSteam, 2019-10-07:
The latest one.

Guilouz, 2019-10-07:
Same issue here, not the same height and layer gauge have stopped to work since latest version.

[image]

OllisGit, 2019-10-07:
Hi all,
the  hight is delivered wrong by the DisplayLayerProgress-Plugin!

Reason: The default height-evaluation is done by looking for the max z-value. 

In your case this line is used:

`M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2`

Please change the Hight-Evaluation method to `Max Z-Value with Extrusion on Z-Layer` in the DisplayLayerProgress-Settings.
The result is 97.90mm

StefanCohen, 2019-10-08:
@ManuGithubSteam I can't reproduce your layer problem with your gcode file:

[image]
The Layer count stays consistent with DisplayLayerProgress during the entire print. The Total Height mode in DisplayLayerprogress was set to "Max Z-Value with Extrusion on Z-Layer" prior to upoading the gcode file. This was tested in Dashboard 1.10.1 with DisplaylayerProgress 1.14.0

StefanCohen, 2019-10-08:
@OllisGit There may be a problem in the latest version of DisplayLayerProgress. I don't do any re-calculations of what DisplayLayerProgress propagates thru the event handler so what we see in the dashboard is what is sent by DisplayLayerProgress. I think 1.14.0 sends the previous layer height instead of the current Layer height. You can see that marked in red in the screenshot above. 

I added a log message in the Dashboard event handler like below and it consistently gets the previous layer height:

```
def on_event(self, event, payload):
  if event == "DisplayLayerProgress_layerChanged" or event == "DisplayLayerProgress_fanspeedChanged":
    self._logger.info("Current Height: " + payload.get('currentHeight'))

```  
This is not specific to this gcode file. I see it on others too.
Can you please confirm if this is the case or not?

OllisGit, 2019-10-08:
Hi @StefanCohen ,
I could reproduce this issue. I debugged my/you code and found two thinks:

- You are only listing for fanspeed- and layerChanged, but not for heightChanged.....sorry, was not documented!!! I add `or event == "DisplayLayerProgress_heightChanged"` and now it is in sycn.
- Sometimes an exception occurs and no information is send to the dashboard:
```
__init__.py", line 102, in on_event
    if int(payload.get('lastLayerDurationInSeconds')) > 0:
ValueError: invalid literal for int() with base 10: '-'
```
Reason: `lastLayerDurationInSeconds` is not a number! Possible solution: Check type and/or try catch

Should solve this issue ;-)

StefanCohen, 2019-10-08:
Thanks!

I noticed that exception this morning and had already fixed it but that undocumented event solved it. ;) A fix is on the way. I'm just a bit puzzled that it worked at all up until 0.14 since I never had heightChanged in the dashboard. But oh well. It should work from the next release anyway.

On a side-note: That gcode file is pretty strange. Did you see how it changes layer height during the layer? Almost as if it is doing mesh levelling but from the slicer instead of in firmware. @ManuGithubSteam Do you know the reason for that?

mdaneman, 2019-10-08:
The issue with DisplayLayerProgress showing 1000mm is that it looks at non printing Z moves (like the z-axis move up once the print is done for example).  To have it show the correct height, change the option for DisplayLayerProgress that says something like "use only extruding moves for layer height calculation" (sorry don't have access to octoprint right now, so can't check the exact name.

StefanCohen, 2019-10-08:
Fixed in v1.10.2

closed 2019-10-08 by StefanCohen

ManuGithubSteam, 2019-10-11:
This is stil present in 1.10.2.
[image]

I reuploaded the gcode after the update.
Reboot/restart after install was not successfull i rebootet by hand...

StefanCohen, 2019-10-12:
I can't reproduce it using the gcode you uploaded before:

[image]

Please upload the one from you screenshot.

ManuGithubSteam, 2019-10-13:
Here you go. Code and screenshot.
[TevoCube3_45m_Prusament PETG - 0.20mm QUALITY - MK3S - Volcano 0.4.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/3721594/TevoCube3_45m_Prusament.PETG.-.0.20mm.QUALITY.-.MK3S.-.Volcano.0.4.zip)
[image]
As you can see: the layer count is way highter than it should be also the height is not correct....

5 layers x 0.2 = 1mm not 5.60

StefanCohen, 2019-10-13:
@ManuGithubSteam I can't reproduce it. This is how it looks on my system at layer 5: 

[image]

My DisplayLayerProgress settings:

Layer Indicator:
```
1		[;\s*LAYER:\s*([0-9]+).*]		CURA
1		[; layer ([0-9]+),.*]		Simplify3D
1		[;LAYER:([0-9]+).*]		ideaMaker
count	[; BEGIN_LAYER_OBJECT.*]		KISSlicer
count	[;BEFORE_LAYER_CHANGE]		Slic3r
```

Total Height Method: Max Z-Value with Extrusion on Z-Layer

Layer Offset: 1 (since I use Cura most of the time)

ManuGithubSteam, 2019-10-13:
Hmm is the layer graph in your screenshot the new normal ? If so, i think there went something wrong with the update as octoprint did not restart as it should be and i had to to it by hand...

StefanCohen, 2019-10-13:
That layer graph was added in 1.10.0. It is enabled in the settings using the "Show Layer Duration Graph" checkbox. Can you see that setting?

ManuGithubSteam, 2019-10-13:
Yes i see it

OllisGit, 2019-10-14:
@ManuGithubSteam 
I uploaded your file in my dev-environment and I can also not reproduce this issue.

Bye the way, I think you are not using the "DisplayLayerProgress-Plugin" to visualize the layer and height in the left side.
You are using "LayerDisplay" and "DisplayZ".

[image]
The red marked box show "DisplayLayerProgress".

This has maybe nothing to do with the current problem, but as an advice you should only use "DisplayLayerProgress", because the Dashboard-Plugin only reads this data.

Or is there a reason why do don't use the output of DisplayLayerProgress in the sidebar?

ManuGithubSteam, 2019-10-14:
Hello
Well i dont need 2 outputs there only one is enough and i prefer the one who gives the more accurate layer lines. and the other ones are not reliable for me. But i use the plugin in the sense that it is activated and should do its work in the background. I just dont want to see two of them... 
Will investigate further tomorrow...
Thanks!

nolspeed29, 2020-01-12:
> @ManuGithubSteam I can't reproduce it. This is how it looks on my system at layer 5:
>
> [image]
>
> My DisplayLayerProgress settings:
>
> Layer Indicator:
>
> ```
> 1		[;\s*LAYER:\s*([0-9]+).*]		CURA
> 1		[; layer [...]

I know this is a closed issue, but I was curious if there was a setting to get the layer duration graph to ignore the first layer as shown in your screenshot?  My first layer is very lengthy due to slow movements, which kills the scaling on the graph and makes it pretty useless.  I would be great to be able to have an option to ignore the 1st layer, which might already be the case?