j7126/OctoPrint-Dashboard #83: Better handling of long filenames

StefanCohen, 2019-10-08:
Long filenames are split over several lines. Should scale the size to fit one line instead:

[image]

StefanCohen, 2019-10-08:
@j7126 Do you have any idea how to scale the string dynamically to fit the containing div?

I'm thinking about using an SVG like in the last example [here](https://css-tricks.com/fitting-text-to-a-container/) but that link provides a ton of alternatives and there may be a better method that I'm not aware of. What do you think?

j7126, 2019-10-08:
@StefanCohen I would say that using a svg in most situations would be fine, however in this case the text color if using a svg will not play nice with themes, 

I think that for this situation using viewport width would be much better. 

If you would like I can do this and make a pull request.

j7126, 2019-10-08:
I realised my mistake that using viewport width will actually not be a good idea and will not work because that works for things that take up the whole page, and the dashboard when not in fullscreen mode does not take up the entire screen. 

I think it would be much better in this case to use fitty https://github.com/rikschennink/fitty which was also mentioned in that link and I have used in the past and works well.

StefanCohen, 2019-10-08:
Thanks. Fitty looks like a nice alternative. I will give it a try.

StefanCohen, 2019-10-08:
That worked fine:
[image]

I wonder where something like this should be called from? I temporarily placed it in the self.onTabChange callback handler sice it is called on print start but that doesn't seem right. There is no callback for .onPrintStart so I guess it has to be a new event handler for PrintStarted? What do you think?

StefanCohen, 2019-10-08:
Nvm. I'm just slow. I got it now :)

StefanCohen, 2019-10-08:
Fixed in v1.10.2

closed 2019-10-08 by StefanCohen