j7126/OctoPrint-Dashboard #84: Colors for text and gauges

wengeol2, 2019-10-08:
As a user I would like to change the color of the gauges and the text. For example to match it with my octoprint theme.

I added some options to "themeify" plugin to get the desired result:
check "Enable customisation" and under "Advanced Options" -> "Custom style" add the following new entries:
```
.dashboardGauge   stroke   #fc8003
.dashboardSmall   color    #fc8003
svg text          fill     #fc8003
```

StefanCohen, 2019-10-08:
Thanks for the feedback. 

That's the way I do it and the way i described it on the [Dashboard wiki](https://github.com/StefanCohen/OctoPrint-Dashboard/wiki/Help). 

I view the themeify method as the current best practice for changing the appearance of Octoprint including plugins. It is a bit "Hacky" and not that user friendly, but it at least provides some kind of separation of duties so every plugin doesn't need to provide its own theming.

I think an even better suggestion would be for Themeify (or maybe a new plugin) to provide some mechanism for plugins to hook into that makes it possible to use "external" styles. so things like text color could be configured in one place and used by all plugins that wants to.  

Anyway. I see your point but I don't see adding theme colour settings in the Dashboard as the right way forward. At least not right now when there is a working alternative and until other alternatives has been explored.

I will however bring up the question with the developer community to get more input because I think this is important question. And who knows, maybe this is someone else's (or my) next plugin project?

wengeol2, 2019-10-09:
Hi Stefan, I didn't see the Themeify section before.. must have missed it.

closed 2019-10-09 by wengeol2

StefanCohen, 2019-10-09:
No problem. It used to be in the README.md file but I moved it since it became a bit crowded.

bshin100, 2020-06-02:
Hi I didn't want to start a new issue on this because it's relatively minor. I've tried changing the color of the texts from Themify as instructed in the wiki, however it only ends up applying to the bed temp and fan gauge. For the hotend, it returns to default colors for some reason. I opened up the console and the CSS profile is indeed the old color. 
[image]