j7126/OctoPrint-Dashboard #85: Support a second DHT11/12 sensor for filament dry box

ManuGithubSteam, 2019-10-08:
**Is your feature request related to a problem? 

Problem: I can not remotely monitor my moisture in my filament dry box

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Ideally in form of a user story so that I know why this is requested:

As a Dasboard user, I want to see the moisture and temp of my filament dry box as well as my enclosure temp/ moisture so we need support for two sensors.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The display of the second (and first) sensor should be toggle able in the settings.

**Additional context**
I want to see my moisture in the dry box from afar. I have already wireing going in the box but no way to read it...

StefanCohen, 2019-10-08:
I knew that would come :)

I'll save this for the future. In the mean-time, please vote for [#260](https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/260) As I would strongly prefer if sensors were exposed via an api from someone knowledgeable instead of me just guessing how it works.

ManuGithubSteam, 2019-10-08:
Hi,

Reading that sensor is really easy if I remember correctly you can just
rebase the raw value the adafruit python scripts do this and it's maybe one
short sheet of code. You could just parse the values from say /tmp/dht.txt
and the user needs to provide means to write that file with the values.
There are plenty of examples around how to read that sensor data. I
personally would implement much of it myself and not relay on other
plugins. See the recent layer débâcle. Also at some point users need like 5
plugins just to get dashboard working wich is not ideal in my view...

StefanCohen, 2019-10-09:
I have now decided to remove the existing sensor implementation and instead read temperature and humidity sensor data from the Enclosure plugin going forward.

There are multiple reasons for this:
1. I'm not comfortable with the sensor domain and have less than two hours of experience with it.
2. I will not attempt to support sensor integrations I don't have access to and can verify myself.
3. The current DHT integration has led to more problems with non-RPi users than I'm prepared to support (mostly due to my inexperience but also with people running on semi-compatible RPi clones).
4. The enclosure plugin is mature and supports more sensors than I will be able or willing to support. The integration can be made seamless without any configuration in the Dashboard.

closed 2019-10-09 by StefanCohen