j7126/OctoPrint-Dashboard #87: Feature Request

retikulumx, 2019-10-08:
It would be cool, if you could implement the "Pause" and "Cancel" button. I'm using an old tablet as display for my printer. But sometimes I have to interrupt the print. 
Thanks :-)

StefanCohen, 2019-10-08:
So I guess you are using it in fullscreen mode on the tablet?

I have already added support for pause/cancel, but it is only available via a context menu by clicking on the file name. I guess that is a no-go on a tablet...

retikulumx, 2019-10-09:
Hi.
Yes, exactly. I'm using it in full screen mode.
Would be very comfortable to place 2 buttons directly to the screen.
But maybe the "cancel" button with a "are you really sure" prompt. If that's not possible, the _pause_ button should be enough I guess.

StefanCohen, 2019-10-12:
WIP:

[image]

@j7126 There is a slight problem with the cancel confirmation modal dialog. It is not displayed on top of the fullscreen element so it is hidden until I exit fullscreen. Do you have any ideas how to solve that? I guess it has to be done from js but I haven't found a way yet.

j7126, 2019-10-13:
@StefanCohen the reason that the confirmation dialog is hidden in fullscreen is because since you are just duplicating the default buttons and confirmation dialog which is part of octoprint, this dialog is outside of the dashboard element. Because only the dashboard element is what is fullscreen that is why this is happening.

I will work on a solution.

j7126, 2019-10-13:
@StefanCohen fixed in #94

StefanCohen, 2019-10-13:
Confirmed fixed in #94. Thanks again!

StefanCohen, 2019-10-13:
Fixed in 1.11.0

closed 2019-10-13 by StefanCohen